openbmc/openbmc #1760: generate-squashfs: Support multiple files

anoo1, 2017-06-08:
The generate-squashfs script from the openpower-pnor-code-mgmt repo needs to support multiple input files. And also use the input file names as the output so that the output is not just pnor.squashfs.tar but instead <filename.pnor>.squashfs.tar.

So that for example a call to:
`generate-squashfs witherspoon.pnor witherspoon_update.pnor`
generates a witherspoon.pnor.squashfs.tar and a witherspoon_update.pnor.squashfs.tar output files.

anoo1, 2017-06-09:
Use cases:
`generate-squashfs <file>.pnor`
  generates a <file>.pnor.squashfs.tar file
`generate-squashfs -f <newfilename> <file>.pnor `
  generates a <newfilename> file
`generate-squashfs <file0>.pnor <file1>.pnor`
  generates a <file0>.pnor.squashfs.tar <file1>.pnor.squashfs.tar files
`generate-squashfs -f <newfilename> <file0>.pnor <file1>.pnor `
  various options, we could make that -f is only supported with one pnor file, or it could take multiple newfilenames, or it could be that it creates <newfile> and <newfilename>1 files

anoo1, 2017-06-09:
The requester didn't need multiple file support after all, so we're just adding the change to create the output filename based on the specified filename, and also fixing the --file parameter so that the user doesn't need to specify an output path but instead it defaults to current directory.

rfrandse, 2017-06-09:
<https://gerrit.openbmc-project.xyz/4698> generate-squashfs: output file name improvements
Resolves  #1760 generate-squashfs: Default file name improvements

closed 2017-06-14 by williamspatrick