openbmc/openbmc #1761: fsi1: can't read slave 02:00 -5 error message seen on Witherspoon after doing BMC code update

PavamanSubramaniyam, 2017-06-09:
Not able to power on the Witherspoon Hardware even after obmcutil poweron command.
The machine is installed with the latest OpenBMC daily build image.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-99"
VERSION_ID="v1.99.6-99-gd444774"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-99"
BUILD_ID="v1.99.6"

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-witherspoon-v1.17-25-g4711f01
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-16-g9abb449b9c57
        hostboot-d96098e
        linux-4.11.2-openpower1-p8f7ea46
        petitboot-v1.4.2-pb361a08
        machine-xml-96746e9
        occ-bc34ad5
        hostboot-binaries-7629121
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%
```

PavamanSubramaniyam, 2017-06-09:
[dmesg.txt](https://github.com/openbmc/openbmc/files/1063815/dmesg.txt)
[journalctl.txt](https://github.com/openbmc/openbmc/files/1063816/journalctl.txt)

PavamanSubramaniyam, 2017-06-09:
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

root@witherspoon:~# obmcutil poweron
root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
```

PavamanSubramaniyam, 2017-08-28:
I am still seeing this issue on the latest Open BMC Firmware images.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.9-57"
VERSION_ID="v1.99.9-57-g8aa1141"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.9-57"
BUILD_ID="v1.99.9"

root@witherspoon:~# dmesg
[ 2122.630000]  slave@00:00: status: 0x84000510, sisc: 0x01040108
[ 2122.630000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2122.630000]  fsi1: can't read slave 00:00 -5
[ 2122.630000] occ-hwmon occ2-dev0: failed to get OCC poll response: -19
[ 2123.530000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.530000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.530000]  slave@01:00: status: 0x04000510, sisc: 0x01040000
[ 2123.530000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.530000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.600000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.630000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.630000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.650000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.660000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.670000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.690000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.720000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.730000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.740000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.750000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.760000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2123.770000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.790000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.810000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.840000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.840000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.870000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.870000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.910000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.910000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.960000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2123.970000]  slave@01:00: status: 0x04000500, sisc: 0x00040000
[ 2124.050000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.050000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.090000]  fsi1: can't read slave 02:00 -5
[ 2124.110000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.110000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.130000]  fsi1: can't read slave 03:00 -5
[ 2124.160000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.160000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.160000]  fsi1: can't read slave 04:00 -5
[ 2124.190000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.190000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.190000]  fsi1: can't read slave 05:00 -5
[ 2124.240000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.250000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.250000]  fsi1: can't read slave 06:00 -5
[ 2124.290000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.290000]  slave@00:00: status: 0x84000500, sisc: 0x00040108
[ 2124.310000]  fsi1: can't read slave 07:00 -5
```

gtmills, 2018-08-20:
@PavamanSubramaniyam Still a problem ? This is a really old driver. Thanks!

closed 2018-08-22 by shenki