openbmc/openbmc #1765: Witherspoon: Power on failed due to /usr/sbin/phosphor-fan-presence-tach service failure

sivassrr, 2017-06-09:
Problem: 
With today's driver [802](https://openpower.xyz/job/openbmc-build/802/) , System failed to power on due to phosphor-fan-control. 

Firmware Details as follows:
```
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-99"
VERSION_ID="v1.99.6-99-gd444774"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-99"
BUILD_ID="v1.99.6"
root@witherspoon:# cat /etc/timestamp
20170609043017
root@witherspoon:# /usr/sbin/pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.17_1.20
	op-build-v1.16-87-g5e69953-dirty
	buildroot-2017.02.2-7-g23118ce
	skiboot-05b8834-p7542d9b
	hostboot-d02273f
	linux-4.11.1-openpower1-p77ac250
	petitboot-v1.4.2-pf0e80a2
	machine-xml-e536539
	occ-8a7df03
	hostboot-binaries-d8f4ab7
	capp-ucode-9c73e9f
	sbe-d770027
[==================================================] 100%
root@witherspoon:# 
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:# obmcutil poweron
root@witherspoon:# obmcutil power
pgood = 0
state = 1
pgood_timeout = 10
root@witherspoon:#

root@witherspoon:# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:# obmcutil power
```

sivassrr, 2017-06-09:
[wsbmc006jounralctlDr802.txt](https://github.com/openbmc/openbmc/files/1064184/wsbmc006jounralctlDr802.txt)

sivassrr, 2017-06-09:
``` 
**Failure Snippet**
Apr 20 19:02:55 witherspoon systemd[1626]: phosphor-fan-presence-tach@0.service: Failed at step EXEC spawning /usr/sbin/phosphor-fan-presence-tach: No such file or directory
Apr 20 19:02:56 witherspoon systemd[1625]: phosphor-fan-control-init@0.service: Failed at step EXEC spawning /usr/sbin/phosphor-fan-control: No such file or directory
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Main process exited, code=exited, status=203/EXEC
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Unit entered failed state.
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Failed with result 'exit-code'.
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Main process exited, code=exited, status=203/EXEC
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Unit entered failed state.
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Failed with result 'exit-code'.
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-control-init@0.service: Service hold-off time over, scheduling restart.
Apr 20 19:02:56 witherspoon systemd[1]: phosphor-fan-presence-tach@0.service: Service hold-off time over, scheduling restart.
```

msbarth, 2017-06-09:
Looks like c50a8c4609786980960235b9f9913ea4dd125176 is where this was introduced.
Working with @bjwyman to determine root cause and fix

closed 2017-06-12 by williamspatrick