openbmc/openbmc #1771: phosphor-settingsd : implement settings validation

dkodihal, 2017-06-11:
Implement methods to validate various settings. The validation must take place when a user changes a setting from the default to a new value.

dkodihal, 2017-06-15:
- [ ] Specify that a setting needs validation in the settings yaml, specify the "type" of validation required. Currently we would need a "range" based validation and a "custom" validation (for network settings).
- [ ] Generate code such that when a "set" operation happens on a setting, the corresponding validator is invoked.
- [ ] Implement the range based validation method. Throw an exception back over the bus when an invalid value is provided by a setting.

dhruvibm, 2017-07-21:
Started working from today

dhruvibm, 2017-07-24:
# **STATUS**
07/24 - Started word from today.

rfrandse, 2017-07-26:
<https://gerrit.openbmc-project.xyz/5491> Validation for settings while setting values.
Resolves  #1771 phosphor-settingsd : implement settings validation

closed 2017-07-29 by williamspatrick