openbmc/openbmc #1772: phosphor-logging : improve metadata provision

dkodihal, 2017-06-11:
As of now, an application with local errors and metadata must have a Makefile/manually generated metadata header (elog-errors.hpp) to get past CI. This is because CI will not do a bitbake (which would have built certain native recipes to generate the metadata header).

This can be simplified for example by hiding the elog-errors.hpp generation command behind an autotools rule. There could be other ways to achieve what is desired here.

zahrens, 2017-07-14:
@devenrao Task list?

devenrao, 2017-08-24:
- [ ] Add a new autoconf macro to phosphor-logging that facilitates in generating elog-errors.hpp by using the parser and mako script files.
- [ ] Modify open-power-debug-collector repository to use the auoconf macro to auto generate elog-errors.hpp at compile time.
- [ ] Modify phosphor-host-ipmid to use autoconf macro to auto-generate elog-errors.hpp. 

1. Need to cater for both newly added errors and using existing errors.

rfrandse, 2017-09-29:
<https://gerrit.openbmc-project.xyz/6966> Generate local elog-errors header file
Resolves  #1772 phosphor-logging : improve metadata provision

sivassrr, 2017-10-04:
Based on discussion in today's scrum, this wouldnt impact FVT TCs.