openbmc/openbmc #1775: phosphor-fan-control: Error in mapper response for getting service name

gkeishin, 2017-06-12:
Driver Info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-116"
VERSION_ID="v1.99.6-116-gb871cc8"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-116"
BUILD_ID="v1.99.6"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Mon Jun 12 15:21:48 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

```
Apr 20 17:33:41 witherspoon phosphor-fan-control[1277]: Error in mapper response for getting service name
Apr 20 17:33:41 witherspoon phosphor-fan-control[1277]: Error in mapper response for getting service name
Apr 20 17:33:41 witherspoon phosphor-fan-control[1277]: Error in mapper response for getting service name
Apr 20 17:33:41 witherspoon phosphor-fan-control[1277]: Error in mapper response for getting service name

```

- [x] Journald log 
[journald_v1.99.6-116-gb871cc8_mapper_error.txt](https://github.com/openbmc/openbmc/files/1068946/journald_v1.99.6-116-gb871cc8_mapper_error.txt)

spinler, 2017-06-14:
The problem here is that fan-control is trying to read the Functional property from a fan in the inventory before fan-monitor has written it, due  to a race condition.  I'm going to fix this race by changing the fan-monitor service to be of Type=notify, and then have it call sd_event_notify after it has initialized so that fan-control will then be started.

rfrandse, 2017-06-15:
<https://gerrit.openbmc-project.xyz/4797> Add type=notify to fan monitor service
Resolves  #1775 phosphor-fan-control: Error in mapper response for getting service name

closed 2017-06-26 by williamspatrick