openbmc/openbmc #1776: BMC fails to boot if RW partition is full

rlippert, 2017-06-12:
Noticed this error on reboot when someone accidentally copied a large file to the BMC writable filesystem taking up all the space:

```
ffs2: notice: (494) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
mkdir: can't create directory 'run/initramfs/rw/work': No space left on device
overlayfs: failed to resolve 'run/initramfs/rw/work': -2
mount: mounting cow on /root failed: No such file or directory
chroot: can't execute '/bin/sh': No such file or directory

Unable to confirm /sbin/init is an executable non-empty file
in merged file system mounted at /root.

Change Root test failed!  Invoking emergency shell.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):
```

Removing the large file from the emergency shell allowed the machine to boot.

williamspatrick, 2017-06-13:
We are going to be moving away from the overlay mount of a jffs2 volume and towards bind-mounts at specific locations with ubifs.  I don't think we will focus on this now, but we should confirm this scenario is handled better when we enable the ubifs path.

anoo1, 2018-01-03:
With the ubifs implementation, the BMC is able to boot if the RW partition is full. Network connection is not affected so the user can ssh to the BMC to delete files and make space.
The BMC stays at NotReady state until space is made, at which point services like the journald will start normally.

closed 2018-01-03 by anoo1