openbmc/openbmc #1777: IPMID typo print logging in journald

gkeishin, 2017-06-12:
Looks like this code may need to fix a bit in print
 
https://github.com/openbmc/phosphor-host-ipmid/blob/2b4e459867d6b6693676c483958ec899be140316/chassishandler.cpp#L1073
```
Apr 20 17:36:46 witherspoon ipmid[1695]: BootPolicy is[ONEsscanf Failed in extracting mac address.
```
Typo `[ONEsscanf`

From
```
printf("BootPolicy is[%s]", p);
```
To
```
printf("BootPolicy is [%s]\n", p);
```

but the char pointer p holding the boot string seems garble up giving ONE instead of ONETIME in this case.. probably kind of looks like it may not be null terminated too

gkeishin, 2017-06-12:
The other thing I notice is that when this happens the OS boot stopped at petitboot.. but there is a bug already opened for it..  This one to address the typo and have a valid error flagged in the print instead of half cooked print.

```
 Petitboot (v1.4.2-p0fd2665)                  8375-22B         0000000000000000
 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq

  System information
  System configuration
  System status log
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell

 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
 Enter=accept, e=edit, n=new, x=exit, l=language, g=log, h=help
 [enP5p1s0f1] Configuring with DHCP
```
- [x] Journald log 
[journald_v1.99.6-116-gb871cc8_boot_print_typo.txt](https://github.com/openbmc/openbmc/files/1069099/journald_v1.99.6-116-gb871cc8_boot_print_typo.txt)

williamspatrick, 2017-06-13:
@gkeishin - Can you just fix the typo? :smile:

gkeishin, 2017-06-14:
Sure thing @williamspatrick

gkeishin, 2017-06-14:
https://gerrit.openbmc-project.xyz/#/c/4778/

rfrandse, 2017-06-14:
<https://gerrit.openbmc-project.xyz/4778> Fix space and newline for boot policy print
Resolves  #1777 IPMID typo print logging in journald

closed 2017-06-16 by williamspatrick