openbmc/openbmc #1779: openpower-occ-control[920]: ObjectMapper GetSubTreePaths failed

PavamanSubramaniyam, 2017-06-13:
I have installed the latest Open BMC and PNOR firmware images on witherspoon hardware.
In the journalctl logs we keep on getting the below messages:

```
Jun 13 06:18:37 witherspoon systemd[1]: Started Phosphor BIOS Code Update.
Jun 13 06:18:38 witherspoon openpower-occ-control[920]: [[0;1;31mObjectMapper GetSubTreePaths failed[[0m
Jun 13 06:18:38 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Main process exited, code=exited, status=255/n/a[[0m
Jun 13 06:18:38 witherspoon systemd[1]: [[0;1;31mFailed to start OpenPOWER OCC controller.[[0m
Jun 13 06:18:38 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Unit entered failed state.[[0m
Jun 13 06:18:38 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Failed with result 'exit-code'.[[0m
Jun 13 06:18:39 witherspoon systemd[1]: org.open_power.OCC.PassThrough.service: Service hold-off time over, scheduling restart.
Jun 13 06:18:39 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Jun 13 06:18:39 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Jun 13 06:18:40 witherspoon settings_manager.py[909]: Running HostSettingsService
Jun 13 06:18:40 witherspoon systemd[1]: Started Phosphor Settings Manager.
Jun 13 06:18:40 witherspoon systemd[1]: Starting Wait for /org/openbmc/settings/host0...
Jun 13 06:18:40 witherspoon systemd[1]: Started Assert bmc_booted LED.
Jun 13 06:18:40 witherspoon openpower-occ-control[936]: [[0;1;31mObjectMapper GetSubTreePaths failed[[0m
Jun 13 06:18:40 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Main process exited, code=exited, status=255/n/a[[0m
Jun 13 06:18:40 witherspoon systemd[1]: [[0;1;31mFailed to start OpenPOWER OCC controller.[[0m
Jun 13 06:18:40 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Unit entered failed state.[[0m
Jun 13 06:18:40 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Failed with result 'exit-code'.[[0m
Jun 13 06:18:40 witherspoon systemd[1]: org.open_power.OCC.PassThrough.service: Service hold-off time over, scheduling restart.
Jun 13 06:18:40 witherspoon systemd[1]: Stopped OpenPOWER OCC controller.
Jun 13 06:18:40 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
```

PavamanSubramaniyam, 2017-06-13:
The machine is installed with the following firmware images.
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-118"
VERSION_ID="v1.99.6-118-g786908a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-118"
BUILD_ID="v1.99.6"

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-witherspoon-v1.17-42-g165d343
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-64-g700b8962e5fe
        hostboot-0f8d0c0
        linux-4.11.2-openpower1-p8f7ea46
        petitboot-v1.4.2-pb361a08
        machine-xml-c5e17e0
        occ-852d3b9
        hostboot-binaries-7629121
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%
```

williamspatrick, 2017-06-13:
Duplicate of #1450.

williamspatrick, 2017-06-13:
You probably should not be testing out of master unless you have some specific feature you are looking for.  We have official drivers internally for Witherspoon machines that have gone through additional stability testing so you will not have to do needless debug like this.

closed 2017-06-13 by williamspatrick