openbmc/openbmc #1781: IPMITOOL SOL console is failing to connect to witherspoon machine

PavamanSubramaniyam, 2017-06-13:
I am trying to connect to the witherspoon hardware using ipmitool sol console command.

```
[root@lop832 src]# ./ipmitool -I lanplus -H ltc-wspoon4-bmc.aus.stglabs.ibm.com -P 0penBmc sol activate
Info: SOL payload already active on another session
[root@lop832 src]# ./ipmitool -I lanplus -H ltc-wspoon4-bmc.aus.stglabs.ibm.com -P 0penBmc sol deactivate
Error de-activating SOL payload: Unspecified error
```

In the journalctl we could see the below error messsages.

```
root@witherspoon:~# journalctl | tail -100
Jun 13 09:43:44 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:43:44 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:43:44 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:43:57 witherspoon systemd[1]: Started SSH Per-Connection Server (9.40.193.27:57506).
Jun 13 09:43:57 witherspoon dropbear[2739]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jun 13 09:43:57 witherspoon dropbear[2739]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jun 13 09:43:57 witherspoon dropbear[2739]: Child connection from ::ffff:9.40.193.27:57506
Jun 13 09:44:00 witherspoon dropbear[2739]: pam_unix(dropbear:auth): authentication failure; logname= uid=0 euid=0 tty=ssh ruser= rhost=  user=root
Jun 13 09:44:02 witherspoon dropbear[2739]: pam_authenticate() failed, rc=7, Authentication failure
Jun 13 09:44:02 witherspoon dropbear[2739]: Bad PAM password attempt for 'root' from ::ffff:9.40.193.27:57506
Jun 13 09:44:05 witherspoon dropbear[2739]: PAM password auth succeeded for 'root' from ::ffff:9.40.193.27:57506
Jun 13 09:44:05 witherspoon dropbear[2740]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Jun 13 09:44:05 witherspoon dropbear[2740]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Jun 13 09:44:24 witherspoon netipmid[1390]: IPMI GROUP EXTENTIONS
Jun 13 09:44:24 witherspoon netipmid[1390]: >> closeSession
Jun 13 09:44:24 witherspoon netipmid[1390]: << closeSession
Jun 13 09:44:24 witherspoon netipmid[1390]: >> GetChannelCapabilities
Jun 13 09:44:24 witherspoon netipmid[1390]: << GetChannelCapabilities
Jun 13 09:44:24 witherspoon netipmid[1390]: >> openSession
Jun 13 09:44:24 witherspoon netipmid[1390]: << openSession
Jun 13 09:44:24 witherspoon netipmid[1390]: >> RAKP12
Jun 13 09:44:24 witherspoon netipmid[1390]: << RAKP12
Jun 13 09:44:24 witherspoon netipmid[1390]: >> RAKP34
Jun 13 09:44:24 witherspoon netipmid[1390]: << RAKP34
Jun 13 09:44:24 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:44:24 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:44:24 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:44:28 witherspoon netipmid[1390]: IPMI GROUP EXTENTIONS
Jun 13 09:44:28 witherspoon netipmid[1390]: >> closeSession
Jun 13 09:44:28 witherspoon netipmid[1390]: << closeSession
Jun 13 09:44:28 witherspoon netipmid[1390]: >> GetChannelCapabilities
Jun 13 09:44:28 witherspoon netipmid[1390]: << GetChannelCapabilities
Jun 13 09:44:28 witherspoon netipmid[1390]: >> openSession
Jun 13 09:44:28 witherspoon netipmid[1390]: << openSession
Jun 13 09:44:28 witherspoon netipmid[1390]: >> RAKP12
Jun 13 09:44:28 witherspoon netipmid[1390]: << RAKP12
Jun 13 09:44:28 witherspoon netipmid[1390]: >> RAKP34
Jun 13 09:44:28 witherspoon netipmid[1390]: << RAKP34
Jun 13 09:44:28 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:44:28 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:44:28 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:44:28 witherspoon netipmid[1390]: Invalid retrieval option
Jun 13 09:45:02 witherspoon netipmid[1390]: IPMI GROUP EXTENTIONS
Jun 13 09:45:02 witherspoon netipmid[1390]: >> closeSession
Jun 13 09:45:02 witherspoon netipmid[1390]: << closeSession
Jun 13 09:45:02 witherspoon netipmid[1390]: >> GetChannelCapabilities
Jun 13 09:45:02 witherspoon netipmid[1390]: << GetChannelCapabilities
Jun 13 09:45:02 witherspoon netipmid[1390]: >> openSession
Jun 13 09:45:02 witherspoon netipmid[1390]: << openSession
Jun 13 09:45:02 witherspoon netipmid[1390]: >> RAKP12
Jun 13 09:45:02 witherspoon netipmid[1390]: << RAKP12
Jun 13 09:45:02 witherspoon netipmid[1390]: >> RAKP34
Jun 13 09:45:02 witherspoon netipmid[1390]: << RAKP34
Jun 13 09:45:02 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:45:02 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:45:02 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:45:02 witherspoon netipmid[1390]: Invalid retrieval option
Jun 13 09:45:09 witherspoon netipmid[1390]: IPMI GROUP EXTENTIONS
Jun 13 09:45:09 witherspoon netipmid[1390]: >> closeSession
Jun 13 09:45:09 witherspoon netipmid[1390]: << closeSession
Jun 13 09:45:09 witherspoon netipmid[1390]: >> GetChannelCapabilities
Jun 13 09:45:09 witherspoon netipmid[1390]: << GetChannelCapabilities
Jun 13 09:45:09 witherspoon netipmid[1390]: >> openSession
Jun 13 09:45:09 witherspoon netipmid[1390]: << openSession
Jun 13 09:45:09 witherspoon netipmid[1390]: >> RAKP12
Jun 13 09:45:09 witherspoon netipmid[1390]: << RAKP12
Jun 13 09:45:09 witherspoon netipmid[1390]: >> RAKP34
Jun 13 09:45:09 witherspoon netipmid[1390]: << RAKP34
Jun 13 09:45:09 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:45:09 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:45:09 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:50:43 witherspoon netipmid[1390]: IPMI GROUP EXTENTIONS
Jun 13 09:50:43 witherspoon netipmid[1390]: >> closeSession
Jun 13 09:50:43 witherspoon netipmid[1390]: << closeSession
Jun 13 09:50:43 witherspoon netipmid[1390]: >> GetChannelCapabilities
Jun 13 09:50:43 witherspoon netipmid[1390]: << GetChannelCapabilities
Jun 13 09:50:43 witherspoon netipmid[1390]: >> openSession
Jun 13 09:50:43 witherspoon netipmid[1390]: << openSession
Jun 13 09:50:43 witherspoon netipmid[1390]: >> RAKP12
Jun 13 09:50:43 witherspoon netipmid[1390]: << RAKP12
Jun 13 09:50:43 witherspoon netipmid[1390]: >> RAKP34
Jun 13 09:50:43 witherspoon netipmid[1390]: << RAKP34
Jun 13 09:50:43 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:50:43 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:50:43 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:50:47 witherspoon netipmid[1390]: IPMI GROUP EXTENTIONS
Jun 13 09:50:47 witherspoon netipmid[1390]: >> closeSession
Jun 13 09:50:47 witherspoon netipmid[1390]: << closeSession
Jun 13 09:50:47 witherspoon netipmid[1390]: >> GetChannelCapabilities
Jun 13 09:50:47 witherspoon netipmid[1390]: << GetChannelCapabilities
Jun 13 09:50:47 witherspoon netipmid[1390]: >> openSession
Jun 13 09:50:47 witherspoon netipmid[1390]: << openSession
Jun 13 09:50:47 witherspoon netipmid[1390]: >> RAKP12
Jun 13 09:50:47 witherspoon netipmid[1390]: << RAKP12
Jun 13 09:50:47 witherspoon netipmid[1390]: >> RAKP34
Jun 13 09:50:47 witherspoon netipmid[1390]: << RAKP34
Jun 13 09:50:47 witherspoon netipmid[1390]: >> setSessionPrivilegeLevel
Jun 13 09:50:47 witherspoon netipmid[1390]: << setSessionPrivilegeLevel
Jun 13 09:50:47 witherspoon netipmid[1390]: E> Table:: Command Not found: 0xb03e
Jun 13 09:50:47 witherspoon netipmid[1390]: Invalid retrieval option
```

PavamanSubramaniyam, 2017-06-13:
[journalctl.txt](https://github.com/openbmc/openbmc/files/1070982/journalctl.txt)
[dmesg.txt](https://github.com/openbmc/openbmc/files/1070984/dmesg.txt)

tomjoseph83, 2017-07-27:
# **STATUS**
7/27 - 

Started working on a fix for this issue. 
Completed the peer review for 1489  
Reviewed patches for 1608.
Debugged the issue with the steps provided in this defect and updated the defect.
 https://bugzilla.linux.ibm.com/show_bug.cgi?id=156592

The plan for tomorrow is to give a patch for this issue and complete the review for 1899 and 1931.

rfrandse, 2017-08-01:
<https://gerrit.openbmc-project.xyz/5667> sol: Handle deactivate payload correctly when session is closed
Resolves  #1781 IPMITOOL SOL console is failing to connect to witherspoon machine

tomjoseph83, 2017-08-07:
# **STATUS**

8/7
Ready for merge review.

closed 2017-08-14 by williamspatrick