openbmc/openbmc #1783: add gpio-keys entries to witherspoon DT for power supply presence

bradbishop, 2017-06-13:


cbostic, 2017-06-13:
Task list:

- [ ]  Identify Port and pin number for each of the new keys

- [ ]  Identify signal polarity for each of the new keys (ACTIVE_HIGH, ... )

- [ ]  Code, test and submit for review

cbostic, 2017-07-13:
Status:
Yesterday - No time spent on this story, focus was on 1727
Today - Will shift focus back to this story and implement polling function.

geissonator, 2017-07-13:
Need to update task list to include new poll function

cbostic, 2017-07-14:
Status Update
July 14, 2017
-------------------------------------------------------
Yesterday
* Worked with Gunnar to test the direct gpio portion of this story.   This is functional and I've sent this up for review to OpenBMC.
* Started work on polling GPIO.  Design complete and coding mostly done.
Today
* Finish up coding for polling GPIO and unit test

mdmillerii, 2017-07-14:
Are you using gpio-keys-polled ?   from `Documentation/gpio/drivers-on-gpio.txt`
```
- gpio-keys-polled: drivers/input/keyboard/gpio_keys_polled.c is used when your
  GPIO line cannot generate interrupts, so it needs to be periodically polled
  by a timer.
```

cbostic, 2017-07-14:
>
> Are you using gpio-keys-polled ? from 
> |Documentation/gpio/drivers-on-gpio.txt|
>
> |- gpio-keys-polled: drivers/input/keyboard/gpio_keys_polled.c is used 
> when your GPIO line cannot generate interrupts, so it needs to be 
> periodically polled by a timer.|

|Hi Milton,

gpio-keys-polled isn't part of this submission - that will come next.  
The particular GPIOs here are direct connected to the AST2400/2500 only.
|

bjwyman, 2017-08-11:
->  ARM: dts: aspeed: witherspoon: Add gpio-keys-polled for fan presence 
--> https://github.com/openbmc/linux/commit/5ea0608616b939bee5f0856a4227efedcc774249
->  kernel: fsi fixes for r/r 
--> https://github.com/openbmc/openbmc/commit/3f6eb4cf07f1ca218dcc8189726f4aaecc7cec13

geissonator, 2017-08-15:
Done

closed 2017-08-15 by geissonator