openbmc/openbmc #1785: Persist host and chassis state requests

dkodihal, 2017-06-13:
Persist host and chassis state requests, so that they can be restored on BMC reset.

dkodihal, 2017-06-29:
- [ ] Persist the value of the request host state transition using Cereal.
- [ ] When the host state manager starts up, if there is a persistent request host state value, deserialize that using Cereal and execute a state transition based on that request.
- [ ] State recipe : add a depends on Cereal.

geissonator, 2017-06-29:
Some notes from review:
- 2nd task should look at settings for boot policy
- Need to update state services to show dependency on new settingsd application

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5223> Add support for host state persistency
Resolves  #1785 Persist host and chassis state requests

dhruvibm, 2017-07-21:
Got 2 +1s, need merge review

dhruvibm, 2017-07-24:
# **STATUS**
07/24 - Planning to push a rework today.

closed 2017-07-27 by williamspatrick