openbmc/openbmc #1786: Define object specific inventory interfaces

dkodihal, 2017-06-13:
Most of the inventory objects today implement a generic inventory Asset interface. We may need specific interfaces (just empty ones?) to let user code (such as UI code) distinguish inventory objects based on interface (instead of having to do it based on object paths).

dkodihal, 2017-08-07:
This story can also cover changes to the occ-control-app, which for the lack of this, was dealing with object paths to figure out that CPU inventory has been added. Instead, use InterfacesAdded.

dkodihal, 2017-08-07:
PIM has some code to ignore empty interfaces, would need some changes there.

dkodihal, 2017-08-10:
- [ ] Define empty interfaces for inventory items - for eg a CPU interface, a DIMM interface etc.
- [ ] Change pimgen.py to accept empty inventory interfaces.
- [ ] Change the vpd parser apps to denote that inventory items must implement these interfaces.
- [ ] Fix occ control app which figures out CPUs by path to instead watch for a CPU interface being added.

devenrao, 2017-08-14:
# **STATUS**
12/08
Understanding existing parser code
14/08
Analyzing the changes to be done

rfrandse, 2017-09-06:
<https://gerrit.openbmc-project.xyz/6380> inventory - Allow empty interfaces in creation of inventory objects
Resolves  #1786 Define object specific inventory interfaces

closed 2017-09-29 by williamspatrick