openbmc/openbmc #1789: phosphor-hwmon should return 0 when hwmon entry disappears

geissonator, 2017-06-13:
During bringup of the new IR35221 driver we found a race condition between when we would unbind the driver (to run required vcs workarounds) and when systemd would correctly stop the phosphor-hwmon service monitoring the corresponding hwmon entry.

If phosphor-hwmon happened to be reading the hwmon entry when the unbind occurred, we would get this:
```
Jun 08 21:34:23 witherspoon phosphor-hwmon-readd[926]: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error'
Jun 08 21:34:23 witherspoon phosphor-hwmon-readd[926]:   what():  filesystem error: cannot canonicalize: No such file or directory [/sys/class/hwmon/hwmon4] [/]
...
Jun 08 21:34:23 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Main process exited, code=dumped, status=6/ABRT
Jun 08 21:34:23 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit entered failed state.
```

During that same second, we would see systemd trying to stop the service:
```
Jun 08 21:34:23 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
```

The conclusion from Brad and I was that we should just have phosphor-hwmon-readd just return a 0 (and exit gracefully) if the hwmon directory being monitored suddenly disappears.  The application that is using the hwmon provided dbus object will be responsible for logging the appropriate errors if needed.

rfrandse, 2017-06-13:
<https://gerrit.openbmc-project.xyz/4764> Don't exit with error when hwmon dir is removed
Resolves  #1789 phosphor-hwmon should return 0 when hwmon entry disappears

williamspatrick, 2017-06-14:
@geissonator @bradbishop - Is the problem that our service file is not using a BindsTo properly?  Or would there still be a race condition even with that?

bradbishop, 2017-06-15:
@williamspatrick BindsTo is more appropriate than StopWhenUnneeded but I doubt it is any faster so I don't think that will help any.

We aren't using BindsTo because the devpath isn't known to the template.  We are using the OF_PATH sysfs attribute as the template parameter.

I think long term we should use the devpath as the template parameter in all of our launched-via-udev services ( I asked Lennart about using other udev event properties like OF_PATH and he said nah ) and use a helper script to launch the service if something else from the event is really needed.

closed 2017-06-16 by williamspatrick

mdmillerii, 2017-06-16:
> I think long term we should use the devpath as the template parameter in all of our launched-via-udev services ( I asked Lennart about using other udev event properties like OF_PATH and he said nah ) and use a helper script to launch the service if something else from the event is really needed.

We can (should?) make udev rules that make custom device names in our own namespace; they can use any attribute to choose the right device.  Then the device name will be known when we make the system.