openbmc/openbmc #1790: IPL: zaius kernel: sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'

swanman, 2017-06-13:
I happened to notice the following repeated warning messages/backtraces in the journal output during our testing on zaius4 last night with 910.1724.20170612n:

```Jun 07 14:05:12 zaius kernel: w1_master_driver w1_bus_master2: Family 0 for 00.800000000000.8c is not registered.
Jun 07 14:05:12 zaius kernel: ------------[ cut here ]------------
Jun 07 14:05:12 zaius kernel: WARNING: CPU: 0 PID: 376 at /home/bmcbuild/releases/obmc909/Builds/ob909.1723.20170607a/output/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
Jun 07 14:05:12 zaius kernel: sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'
Jun 07 14:05:12 zaius kernel: CPU: 0 PID: 376 Comm: w1_bus_master2 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Jun 07 14:05:12 zaius kernel: Hardware name: ASpeed SoC
Jun 07 14:05:12 zaius kernel: [<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
Jun 07 14:05:12 zaius kernel: [<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
Jun 07 14:05:12 zaius kernel: [<80110f50>] (__warn) from [<80110fb0>] (warn_slowpath_fmt+0x44/0x6c)
Jun 07 14:05:12 zaius kernel: [<80110fb0>] (warn_slowpath_fmt) from [<80200888>] (sysfs_warn_dup+0x50/0x70)
Jun 07 14:05:12 zaius kernel: [<80200888>] (sysfs_warn_dup) from [<80200acc>] (sysfs_do_create_link_sd+0x7c/0x94)
Jun 07 14:05:12 zaius kernel: [<80200acc>] (sysfs_do_create_link_sd) from [<802ffc0c>] (bus_add_device+0x68/0x168)
Jun 07 14:05:12 zaius kernel: [<802ffc0c>] (bus_add_device) from [<802fe5b4>] (device_add+0x380/0x52c)
Jun 07 14:05:12 zaius kernel: [<802fe5b4>] (device_add) from [<80355998>] (w1_attach_slave_device+0x1ec/0x3b8)
Jun 07 14:05:12 zaius kernel: [<80355998>] (w1_attach_slave_device) from [<80355ccc>] (w1_slave_found+0xc0/0xd0)
Jun 07 14:05:12 zaius kernel: [<80355ccc>] (w1_slave_found) from [<803564ac>] (w1_search+0x1d4/0x27c)
Jun 07 14:05:12 zaius kernel: [<803564ac>] (w1_search) from [<803565a0>] (w1_search_process_cb+0x4c/0x108)
Jun 07 14:05:12 zaius kernel: [<803565a0>] (w1_search_process_cb) from [<8035672c>] (w1_process+0x68/0x114)
Jun 07 14:05:12 zaius kernel: [<8035672c>] (w1_process) from [<80127888>] (kthread+0x10c/0x124)
Jun 07 14:05:12 zaius kernel: [<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
Jun 07 14:05:12 zaius kernel: ---[ end trace bd7b4e9b72670585 ]---
Jun 07 14:05:12 zaius kernel: w1_slave_driver 00-800000000000: Device registration [00-800000000000] failed. err=-17
Jun 07 14:05:12 zaius kernel: w1_master_driver w1_bus_master2: w1_attach_slave_device: Attaching 00-800000000000 failed.
Jun 07 14:05:12 zaius kernel[796]: w1_master_driver w1_bus_master2: Family 0 for 00.800000000000.8c is not registered.
Jun 07 14:05:12 zaius kernel[796]: ------------[ cut here ]------------
Jun 07 14:05:12 zaius kernel[796]: WARNING: CPU: 0 PID: 376 at /home/bmcbuild/releases/obmc909/Builds/ob909.1723.20170607a/output/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
Jun 07 14:05:12 zaius kernel[796]: sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'
Jun 07 14:05:12 zaius kernel[796]: CPU: 0 PID: 376 Comm: w1_bus_master2 Not tainted 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Jun 07 14:05:12 zaius kernel[796]: Hardware name: ASpeed SoC
Jun 07 14:05:12 zaius kernel[796]: [<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
Jun 07 14:05:12 zaius kernel[796]: [<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
Jun 07 14:05:12 zaius kernel[796]: [<80110f50>] (__warn) from [<80110fb0>] (warn_slowpath_fmt+0x44/0x6c)
Jun 07 14:05:12 zaius kernel[796]: [<80110fb0>] (warn_slowpath_fmt) from [<80200888>] (sysfs_warn_dup+0x50/0x70)
Jun 07 14:05:12 zaius kernel[796]: [<80200888>] (sysfs_warn_dup) from [<80200acc>] (sysfs_do_create_link_sd+0x7c/0x94)
Jun 07 14:05:12 zaius kernel[796]: [<80200acc>] (sysfs_do_create_link_sd) from [<802ffc0c>] (bus_add_device+0x68/0x168)
Jun 07 14:05:12 zaius kernel[796]: [<802ffc0c>] (bus_add_device) from [<802fe5b4>] (device_add+0x380/0x52c)
Jun 07 14:05:12 zaius kernel[796]: [<802fe5b4>] (device_add) from [<80355998>] (w1_attach_slave_device+0x1ec/0x3b8)
Jun 07 14:05:12 zaius kernel[796]: [<80355998>] (w1_attach_slave_device) from [<80355ccc>] (w1_slave_found+0xc0/0xd0)
Jun 07 14:05:12 zaius kernel[796]: [<80355ccc>] (w1_slave_found) from [<803564ac>] (w1_search+0x1d4/0x27c)
Jun 07 14:05:12 zaius kernel[796]: [<803564ac>] (w1_search) from [<803565a0>] (w1_search_process_cb+0x4c/0x108)
Jun 07 14:05:12 zaius kernel[796]: [<803565a0>] (w1_search_process_cb) from [<8035672c>] (w1_process+0x68/0x114)
Jun 07 14:05:12 zaius kernel[796]: [<8035672c>] (w1_process) from [<80127888>] (kthread+0x10c/0x124)
Jun 07 14:05:12 zaius kernel[796]: [<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
Jun 07 14:05:12 zaius kernel[796]: ---[ end trace bd7b4e9b72670585 ]---
Jun 07 14:05:12 zaius kernel[796]: w1_slave_driver 00-800000000000: Device registration [00-800000000000] failed. err=-17
Jun 07 14:05:12 zaius kernel[796]: w1_master_driver w1_bus_master2: w1_attach_slave_device: Attaching 00-800000000000 failed.
Jun 07 14:05:13 zaius kernel: w1_master_driver w1_bus_master4: Family 0 for 00.800000000000.8c is not registered.
Jun 07 14:05:13 zaius kernel: ------------[ cut here ]------------
Jun 07 14:05:13 zaius kernel: WARNING: CPU: 0 PID: 380 at /home/bmcbuild/releases/obmc909/Builds/ob909.1723.20170607a/output/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
Jun 07 14:05:13 zaius kernel: sysfs: cannot create duplicate filename '/bus/w1/devices/00-800000000000'
Jun 07 14:05:13 zaius kernel: CPU: 0 PID: 380 Comm: w1_bus_master4 Tainted: G        W       4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1
Jun 07 14:05:13 zaius kernel: Hardware name: ASpeed SoC
Jun 07 14:05:13 zaius kernel: [<801073cc>] (unwind_backtrace) from [<8010559c>] (show_stack+0x10/0x14)
Jun 07 14:05:13 zaius kernel: [<8010559c>] (show_stack) from [<80110f50>] (__warn+0xd8/0xf4)
Jun 07 14:05:13 zaius kernel: [<80110f50>] (__warn) from [<80110fb0>] (warn_slowpath_fmt+0x44/0x6c)
Jun 07 14:05:13 zaius kernel: [<80110fb0>] (warn_slowpath_fmt) from [<80200888>] (sysfs_warn_dup+0x50/0x70)
Jun 07 14:05:13 zaius kernel: [<80200888>] (sysfs_warn_dup) from [<80200acc>] (sysfs_do_create_link_sd+0x7c/0x94)
Jun 07 14:05:13 zaius kernel: [<80200acc>] (sysfs_do_create_link_sd) from [<802ffc0c>] (bus_add_device+0x68/0x168)
Jun 07 14:05:13 zaius kernel: [<802ffc0c>] (bus_add_device) from [<802fe5b4>] (device_add+0x380/0x52c)
Jun 07 14:05:13 zaius kernel: [<802fe5b4>] (device_add) from [<80355998>] (w1_attach_slave_device+0x1ec/0x3b8)
Jun 07 14:05:13 zaius kernel: [<80355998>] (w1_attach_slave_device) from [<80355ccc>] (w1_slave_found+0xc0/0xd0)
Jun 07 14:05:13 zaius kernel: [<80355ccc>] (w1_slave_found) from [<803564ac>] (w1_search+0x1d4/0x27c)
Jun 07 14:05:13 zaius kernel: [<803564ac>] (w1_search) from [<803565a0>] (w1_search_process_cb+0x4c/0x108)
Jun 07 14:05:13 zaius kernel: [<803565a0>] (w1_search_process_cb) from [<8035672c>] (w1_process+0x68/0x114)
Jun 07 14:05:13 zaius kernel: [<8035672c>] (w1_process) from [<80127888>] (kthread+0x10c/0x124)
Jun 07 14:05:13 zaius kernel: [<80127888>] (kthread) from [<80102278>] (ret_from_fork+0x14/0x3c)
Jun 07 14:05:13 zaius kernel: ---[ end trace bd7b4e9b72670586 ]---
etc
[... 83 lines cut ...]
state[bmc]: Ready
state[boot_progress]: FW Progress, Starting OS
state[host]: Running
state[os_ping]: 0
state[os_login]: 0
state[os_run_cmd]: 0

FFDC data files:
/gsa/ausgsa/projects/a/autoipl/status/zaius4.170612.113319.001.status
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_dmesg.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_general.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_inventory.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_journalctl.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_meminfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_proc_list.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_procinfo.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.BMC_sensor_list.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.SOL.txt
/gsa/ausgsa/projects/a/autoipl/FFDC/zaius4.170612.141933.core_host_checkstop..910

SOL summary:

========================================================================================== 
```
[zaius4.170612.141933.BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/1072505/zaius4.170612.141933.BMC_journalctl.txt)

GHF, 2017-06-13:
Is this an EVT1 Zaius motherboard?

swanman, 2017-06-14:
I could not tell you ... how can this be checked?

GHF, 2017-06-16:
You can read the board revision through the BMC at GPIOs M3–M7. EVT1 would read those as all zeros, EVT2 as 00001, EVT3 00010.

mdmillerii, 2018-05-31:
@swanman please follow the instructions given June 16th to check your board rev.  This is an expected error on early cards that were missing a pull-up resistor on the w1 buses that are use to read temperatures.

closed 2019-04-26 by stale[bot]