openbmc/openbmc #1791: Witherspoon: Can't format the "alt" bmc chip with kernel 4.10

anoo1, 2017-06-13:
With kernel 4.10.x, if the "alt" bmc chip is erased or formatted (using flash_eraseall or ubiformat), the first 2 bytes are not cleared/written. This works with the previous kernel 4.7.x

```
# flash_eraseall /dev/mtd6
# hexdump -C /dev/mtd6 |more
00000000  55 55 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |UU..............|
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00001000  55 55 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |UU..............|
00001010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
```

```
# ubiformat /dev/mtd6
# hexdump -C /dev/mtd6 |more
00000000  55 55 94 55 42 49 21 01  01 00 05 7f ff ef ff 00  |UU.UBI!.........|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  55 55 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |UU..............|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
```

anoo1, 2017-06-13:
Some example versions:

Version of working system:
```
# uname -a
Linux w42 4.7.10-87950053ee336e99ea0f081695b314c962a0859c #1 Tue Jun 6 14:22:28 CDT 2017 armv6l GNU/Linux
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.6"
VERSION_ID="ibm-v1.99.6-0-r1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.6"
BUILD_ID="ibm-v1.99.6-0-r1"
```

Versions of non-working systems:
```
# uname -a
Linux witherspoon 4.10.17-c694e47cbc1efc2bc181b2ac7137dfda04f974c2 #2 Fri Jun 9 14:03:58 CDT 2017 armv6l GNU/Linux
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-137"
VERSION_ID="v1.99.6-137-g202f4bb"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-137"
BUILD_ID="v1.99.6"

# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Tue Jun 13 09:49:52 UTC 2017 armv6l GNU/Linux
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-119"
VERSION_ID="v1.99.6-119-g33a391e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-119"
BUILD_ID="v1.99.6"
```

amboar, 2017-06-26:
@anoo1 The output above is the same system, same flash chip, but different kernel? If so, we should call in @cdrx.

cdrx, 2017-06-26:
@cdrx is not the user you are looking for

williamspatrick, 2017-06-26:
cdrx should be @legoater

amboar, 2017-06-26:
@cdrx sorry for the random ping, I indeed meant @legoater :)

legoater, 2017-06-27:
could you give access to the system to take a look ?

legoater, 2017-06-27:
I have reproduced on a witherspoon system.

anoo1, 2017-07-12:
hi @legoater , wondering if you have additional insights?

legoater, 2017-07-24:
The driver made the wrong assumption that the 4B setting was autodetected for all chips of the AST2500 FMC flash controller. I sent a fix to @shenki on the mailing list.

rfrandse, 2017-07-26:
<https://gerrit.openbmc-project.xyz/5528> kernel: ftgmac100, smc, dps310 fixes
Resolves:  #1791 Witherspoon: Can't format the "alt" bmc chip with kernel 4.10

closed 2017-07-26 by williamspatrick