openbmc/openbmc #1792: Fan Control - Add control event entries in yaml for witherspoon

msbarth, 2017-06-13:
Once necessary sensors are available, add the following to the events yaml:
- [ ] Add fan floor speed table based on ambient temp sensor(s) (openbmc/openbmc#940)
- [ ] Add fan ceiling speed table based on ambient temp sensor(s) (openbmc/openbmc#1628)

msbarth, 2017-07-11:
Speed increase and decrease base example for fan-control events.yaml update:

groups:
    - name: zone0_cores
      description: Group of core temperature sensors for zone 0
      type: sensors
      members:
          - /temperature/p0_core0_temp
          - /temperature/p0_core1_temp
          - /temperature/p0_core2_temp
          - /temperature/p0_core3_temp
          - /temperature/p0_core4_temp
          - /temperature/p0_core5_temp
          - /temperature/p0_core6_temp
          - /temperature/p0_core7_temp
          - /temperature/p0_core8_temp
          - /temperature/p0_core9_temp
          - /temperature/p0_core10_temp
          - /temperature/p0_core11_temp
          - /temperature/p0_core12_temp
          - /temperature/p0_core13_temp
          - /temperature/p0_core14_temp
          - /temperature/p0_core15_temp
          - /temperature/p0_core16_temp
          - /temperature/p0_core17_temp
          - /temperature/p0_core18_temp
          - /temperature/p0_core19_temp
          - /temperature/p0_core20_temp
          - /temperature/p0_core21_temp
          - /temperature/p0_core22_temp
          - /temperature/p0_core23_temp
          - /temperature/p1_core0_temp
          - /temperature/p1_core1_temp
          - /temperature/p1_core2_temp
          - /temperature/p1_core3_temp
          - /temperature/p1_core4_temp
          - /temperature/p1_core5_temp
          - /temperature/p1_core6_temp
          - /temperature/p1_core7_temp
          - /temperature/p1_core8_temp
          - /temperature/p1_core9_temp
          - /temperature/p1_core10_temp
          - /temperature/p1_core11_temp
          - /temperature/p1_core12_temp
          - /temperature/p1_core13_temp
          - /temperature/p1_core14_temp
          - /temperature/p1_core15_temp
          - /temperature/p1_core16_temp
          - /temperature/p1_core17_temp
          - /temperature/p1_core18_temp
          - /temperature/p1_core19_temp
          - /temperature/p1_core20_temp
          - /temperature/p1_core21_temp
          - /temperature/p1_core22_temp
          - /temperature/p1_core23_temp
actions:
    - name: set_net_increase_speed
      description: Set the speed increase delta based on sensor's value and increase size
      parameters:
          - property
          - delta
    - name: set_net_decrease_speed
      description: Set the speed decrease deleta base on sensor's value and decrease size
      parameters:
          - property
          - delta
events:
    - name: speed_increase_changes_based_on_core_temps
      zone_conditions:
          - name: air_cooled_chassis
            zones:
                - 0
          - name: water_and_air_cooled_chassis
            zones:
                - 0
      group: zone0_cores
      interface: xyz.openbmc_project.Control.FanSpeed
      property:
          name: Target
          type: uint64_t
      action:
          name: set_net_increase_speed
          property: 73000
          delta:
              value: 400
              type: uint64_t
    - name: speed_decrease_changes_based_on_core_temps
      zone_conditions:
          - name: air_cooled_chassis
            zones:
                - 0
          - name: water_and_air_cooled_chassis
            zones:
                - 0
      group: zone0_cores
      interface: xyz.openbmc_project.Control.FanSpeed
      property:
          name: Target
          type: uint64_t
      action:
          name: set_net_decrease_speed
          property: 70000
          delta:
              value: 40
              type: uint64_t

rfrandse, 2017-08-16:
<https://gerrit.openbmc-project.xyz/6032> Add fan speed inc/dec control based on core temps
Resolves  #1792 Fan Control - Add control event entries in yaml for witherspoon

closed 2017-08-21 by williamspatrick