openbmc/openbmc #1793: Witherspoon: IPMID - Failed to find sensor error

sivassrr, 2017-06-14:
Problem:

Failed to find Sensor error from IPMID. 

BMC/PNOR Version details: 
```
VERSION="v1.99.6-130"
VERSION_ID="v1.99.6-130-g35cfa84"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-130"
BUILD_ID="v1.99.6"
root@witherspoon:# cat /etc/timestamp
20170613143859
root@witherspoon:# /usr/sbin/pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.16_1.24
        op-build-v1.16-83-ge67b53c-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-rc2
        hostboot-d02273f
        linux-4.11.1-openpower1-pc11ec03
        petitboot-v1.4.2-pcb6424c
        machine-xml-e536539
        occ-8a7df03
        hostboot-binaries-d8f4ab7
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%
```

Failure Snippet:
```
Apr 20 17:49:37 witherspoon ipmid[944]: Failed to get /system/chassis/motherboard/cpu1 busname: (null)
Apr 20 17:49:37 witherspoon ipmid[944]: A host system event was received
Apr 20 17:50:44 witherspoon system_manager.py[865]: Running System State: HOST_BOOTED
Apr 20 17:51:07 witherspoon host_watchdog.exe[822]: Polling watchdog: 1
Apr 20 17:51:54 witherspoon ipmid[944]: ERROR:[0x80] handling NetFn:[0x2C], Cmd:[0x3]
Apr 20 17:51:54 witherspoon system_manager.py[865]: ERROR SystemManager: 9 not found in lookup
Apr 20 17:51:54 witherspoon ipmid[944]: Failed to get  busname: No such file or directory
Apr 20 17:51:54 witherspoon ipmid[944]: Failed to find Sensor 0x09  <===
Apr 20 17:51:54 witherspoon ipmid[944]: ERROR:[0xCB] handling NetFn:[0x4], Cmd:[0x2D]
Apr 20 17:51:55 witherspoon ipmid[944]: Failed to get /org/openbmc/inventory/system/powerlimit busname: No such file or directory
Apr 20 17:51:55 witherspoon ipmid[944]: Could not find sensor 8
Apr 20 17:51:55 witherspoon system_manager.py[865]: ERROR SystemManager: 9 not found in lookup
Apr 20 17:51:55 witherspoon ipmid[944]:

```
[journalctl814.txt](https://github.com/openbmc/openbmc/files/1073970/journalctl814.txt)

Had a discussion with @dkodihal , Decided to open issue for sensor related failure, and 
ignore "ERROR Unable to get the fru info for FRU=26 & Apr 20 17:49:21 witherspoon ipmid[944]: Error updating inventory". Because those are log and it is getting cleaned up.

gtmills, 2018-05-23:
@sivassrr Still an issue? If not, can you close?

sivassrr, 2018-08-21:
Generally, Good to know there is no issue. But it is always better to isolate how it got fixed etc. 
Hopefully we need to identify RCA. But this time, I agree, there are bunch of changes went in after this issue. So I am closing some of them must have fixed.