openbmc/openbmc #1794: Poweron Failed [Witherspoon]

gkeishin, 2017-06-14:
Driver Info:
```
https://openpower.xyz/job/openbmc-build/816/
Build : v1.99.6-133-g20a9336
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Wed Jun 14 08:22:54 UTC 2017 armv6l GNU/Linux
```

```
Apr 20 17:31:40 witherspoon systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
Apr 20 17:31:40 witherspoon systemd[1]: Stopped Wait for Power0 to turn on.
Apr 20 17:31:40 witherspoon systemd[1]: op-wait-power-on@0.service: Unit entered failed state.
Apr 20 17:31:40 witherspoon systemd[1]: op-wait-power-on@0.service: Failed with result 'signal'.
```
```
● op-wait-power-on@0.service - Wait for Power0 to turn on
   Loaded: loaded (/lib/systemd/system/obmc-chassis-poweron@0.target.requires/../op-wait-power-on@.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2017-04-20 17:31:40 UTC; 8min ago
  Process: 987 ExecStart=/usr/sbin/pgood_wait /org/openbmc/control/power%i on (code=killed, signal=TERM)
 Main PID: 987 (code=killed, signal=TERM)

Apr 20 17:31:38 witherspoon systemd[1]: Starting Wait for Power0 to turn on...
Apr 20 17:31:40 witherspoon systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
Apr 20 17:31:40 witherspoon systemd[1]: Stopped Wait for Power0 to turn on.
Apr 20 17:31:40 witherspoon systemd[1]: op-wait-power-on@0.service: Unit entered failed state.
Apr 20 17:31:40 witherspoon systemd[1]: op-wait-power-on@0.service: Failed with result 'signal'.
```

- [x] Journald log 
[journald_v1.99.6-133-g20a9336_poweron_failed.txt](https://github.com/openbmc/openbmc-test-automation/files/1074092/journald_v1.99.6-133-g20a9336_poweron_failed.txt)

gkeishin, 2017-06-14:
Gerrit build 815/816/817 broken..

gkeishin, 2017-06-14:
Looks like this broke the power on https://github.com/openbmc/openbmc/commit/1fd06ed3f000caab3abb140a1dab155a6f7a2d28

```
● ir35221-off-bind@0.service - Power off mode bind ir35221 device driver devs
   Loaded: loaded (/lib/systemd/system/obmc-chassis-poweroff@0.target.requires/../ir35221-off-bind@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-04-20 17:40:33 UTC; 19min ago
  Process: 1115 ExecStart=/usr/bin/ir35221-unbind-bind.sh bind (code=exited, status=1/FAILURE)
 Main PID: 1115 (code=exited, status=1/FAILURE)

Apr 20 17:40:32 witherspoon systemd[1]: Starting Power off mode bind ir35221 device driver devs...
Apr 20 17:40:33 witherspoon ir35221-unbind-bind.sh[1115]: /usr/bin/ir35221-unbind-bind.sh: line 6: echo: write error: No such device
Apr 20 17:40:33 witherspoon ir35221-unbind-bind.sh[1115]: /usr/bin/ir35221-unbind-bind.sh: line 7: echo: write error: No such device
Apr 20 17:40:33 witherspoon ir35221-unbind-bind.sh[1115]: /usr/bin/ir35221-unbind-bind.sh: line 8: echo: write error: No such device
Apr 20 17:40:33 witherspoon ir35221-unbind-bind.sh[1115]: /usr/bin/ir35221-unbind-bind.sh: line 9: echo: write error: No such device
Apr 20 17:40:33 witherspoon systemd[1]: ir35221-off-bind@0.service: Main process exited, code=exited, status=1/FAILURE
Apr 20 17:40:33 witherspoon systemd[1]: Failed to start Power off mode bind ir35221 device driver devs.
Apr 20 17:40:33 witherspoon systemd[1]: ir35221-off-bind@0.service: Unit entered failed state.
Apr 20 17:40:33 witherspoon systemd[1]: ir35221-off-bind@0.service: Failed with result 'exit-code'
```

gkeishin, 2017-06-14:
```
On boot
root@witherspoon:~# uptime
 17:30:04 up 0 min,  load average: 4.55, 1.22, 0.41
root@witherspoon:~# ls /sys/bus/i2c/drivers/ir35221
bind    uevent  unbind
root@witherspoon:~#

On trying to powering on
root@witherspoon:~# ls /sys/bus/i2c/drivers/ir35221
bind    uevent  unbind
root@witherspoon:~#
```

gkeishin, 2017-06-15:
@geissonator  The fix u added for this one is good to close this one or this still need to be carry forward to work on solution ?

geissonator, 2017-06-15:
Fix went in here, we're good - https://gerrit.openbmc-project.xyz/#/c/4789/

closed 2017-06-15 by geissonator