openbmc/openbmc #1796: how to monitor sensors data

sdemirhan, 2017-06-15:
Hi All,

I use openbmc code for garrison platform. I want to monitor sensors data like temperature sensors data. I understand restful API is only used for managing.  How do i monitor sensors data using Bmc which is coded by openbmc. I can read temperature data using sys file system. Is there  any other way to monitor sensors data or read sensors data? 

Thank you for your time.

agangidi53, 2017-06-16:
@sdemirhan I'm not sure on garrison, but on barreleye you can get the temperature sensor data via REST API as well.

yadron, 2017-06-16:
Try these too commands one after other for retrieving sensor info (and host need to be poweron before that):

```
curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://172.17.27.215/login
# - this for authorization (and session stored to ./cjar file)

curl -b cjar -k https://172.17.27.215/org/openbmc/sensors/temperature/membuf0
# - this for retrieving membuf0 temperature (that is temp of first memory riser).
```

; and the below command output the list of all temp sensors available:

`curl -b cjar -k https://172.17.27.215/org/openbmc/sensors/temperature/`

geissonator, 2017-06-16:
A lot of that dbus sensor info comes from hwmon function.  We have the phosphor-hwmon repo which you can setup hwmon conf files that will provide the input sensors via dbus objects (so you can then read those dbus objects to get the temperatures).

sdemirhan, 2017-06-19:
Hi yadron,
When i wrote the commands you suggest me I got the below error.
--root@ddyhpc-HP-ZBook-15-G3:/home/ddy-hpc# curl -b cjar -k https://10.42.0.220/og/openbmc/sensors/temperature/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /org/openbmc/sensors/temperature"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}
--root@ddyhpc-HP-ZBook-15-G3:/home/ddy-hpc# curl -b cjar -k https://10.42.0.220/org/openbmc/sensors/temperature/membuf0
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /org/openbmc/sensors/temperature/membuf0"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}

Thank you for your time.
Seda

sdemirhan, 2017-06-19:
Hi Geissonator,

Thank you for your interest. Do you have any documents how i can do that you write? 
I do not know steps to provide the input sensors via dbus objects. May you explain which steps i should follow?

Thank you for your time.

Seda

yadron, 2017-06-19:
@sdemirhan , very likely the host was down. If host down, then no temp sensors and "404 Not Found". Please run "obmcutil poweron" and then check with "obmcutil power" that host booted and after that try again.

If still not Ok, then please tell more details (ver that is content of /etc/os-release and if host booted well and other).

Regarding reading sensor from DBUS, see:
https://github.com/openbmc/openbmc/issues/1632

sdemirhan, 2017-06-19:
Thank youuu!!!
It works:)

Seda

sdemirhan, 2017-06-20:
Thank you i upgrade openbmc to v1.99.7.

root@garrison:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-10"
VERSION_ID="v1.99.7-10-g521a594"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-10"
BUILD_ID="v1.99.7"

But when i want to read temperature sensors i did not access the repository. I read temperature sensors using restful after host booted.

curl -b cjar -k https://10.42.0.188/xyz/openbmc_project/Sensors/temperature/
{
"data": {
"description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/Sensors/temperature"
},
"message": "404 Not Found",
"status": "error"
}

curl -b cjar -k https://10.42.0.188/xyz/openbmc_project/Sensors/temperature/ambient
{
"data": {
"description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/Sensors/temperature/ambient"
},
"message": "404 Not Found",
"status": "error"
}

Thank you for your time.

yadron, 2017-06-20:
I think these are a set of new RestAPI bugs due migration of "/org/openbmc/sensors/"  to "/xyz/openbmc_project/Sensors/", because doesn't work at Barreleye (v1.99.7-5) too _(even it worked before)_:

```
yadron:~$ curl -b cjar -k https://172.17.27.28/xyz/openbmc_project/sensors/
{
  "data": [], 
  "message": "200 OK", 
  "status": "ok"
}

yadron:~$ curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/
{
  "data": [], 
  "message": "200 OK", 
  "status": "ok"
}
```

However, if I try all sensors one by one that should be available at Barreleye, then some still accessible using old "org" way:

```
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/ambient
{
  "data": {
    "critical_lower": 0, 
    "critical_upper": 0, 
    "emergency_enabled": 0, 
    "error": 0, 
    "filename": "/sys/class/hwmon/hwmon1/temp1_input", 
    "offset": 0, 
    "scale": -3, 
    "threshold_state": "NORMAL", 
    "thresholds_enabled": 0, 
    "units": "C", 
    "value": 31500, 
    "warning_lower": 0, 
    "warning_upper": 0, 
    "worst_threshold_state": "NORMAL"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

and xyz works too:

```
curl -b cjar -k https://172.17.27.28/xyz/openbmc_project/sensors/temperature/ambient
{
  "data": {
    "Scale": -3, 
    "Unit": "xyz.openbmc_project.Sensor.Value.Unit.DegreesC", 
    "Value": 31500
  }, 
  "message": "200 OK", 
  "status": "ok"
}

```

Let's list all Temps* that accessible _(using program that I've written for retrieving all that could be by curl)_:

```
ls 1497968905.43/FOUND/temperature/
ambient/
rtc/
```

Let's list all that "error" _(so cannot get temp of CPU or DIMM currently, but some other DIMMs are just not available because not inserted)_:

[... 130 lines cut ...]
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/dimm5 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/dimm2 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/dimm3 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/dimm12 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/membuf1 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/dimm15 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/membuf0 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core6 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core7 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core4 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core5 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core10 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core11 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core8 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/cpu0/core9 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/membuf3 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/dimm1 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/membuf2 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/powercap/system_power 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/powercap/min_cap 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/powercap/curr_cap 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/powercap/max_cap 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/powercap/user_cap 2>1
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/powercap/n_cap 2>1
```

sdemirhan, 2017-06-29:
Hi All,
i use v1.99.7 version of openbmc and i want to see sensors data. So that i use restful API. 
when i use older version of openbmc, i can use org/openbmc/sensors/temperature/ command to read temperature data, but at the new version of openbmc, i cannot read sensor data using org/openbmc/sensors/temperature/ and also xyz/openbmc_project/sensors/temperature/. 
commands which i used for reading sensors data is given below. (these result is taken when host is booted.)

----curl -b cjar -k https://10.42.0.187/xyz/openbmc_project/sensors/temperature/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/sensors/temperature"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}

----curl -b cjar -k https://10.42.0.187/org/openbmc/sensors/temperature/enumerate
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /org/openbmc/sensors/temperature"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}
-----curl -b cjar -k https://10.42.0.187/xyz/openbmc_project/
{
  "data": [
    "/xyz/openbmc_project/dump", 
    "/xyz/openbmc_project/control", 
    "/xyz/openbmc_project/object_mapper", 
    "/xyz/openbmc_project/inventory", 
    "/xyz/openbmc_project/logging", 
    "/xyz/openbmc_project/software", 
    "/xyz/openbmc_project/led", 
    "/xyz/openbmc_project/state"
  ], 
  "message": "200 OK", 
  "status": "ok"
}

when i see result with xyz/openbmc_project/  command, i realize that the sensors repository is not present. 

How can i implement sensor repository in dbus? or Do you have any commands why i cannot read sensors data?

If information i give is not enough, please let me know.

Thank you!!!

Seda

yadron, 2017-06-29:
I think that the problem that kernel (linux-obmc) updated from 4.7 to 4.10 (in OpenBMC "v1.99.6-2" from ~ 2 June it was 4.7 Linux kernel and in newer OpenBMC kernel 4.10) and as result some sensors not available anymore for some platforms. The original "ls -l /sys/firmware/devicetree/base/ahb/apb//i2c\@1e78a000/*" probably exists, but other required info lost.

What is the result of these two cmds:

```
ls /sys/class/

cat /sys/class/hwmon/*/name
```

and are there any messages like

```
journalctl |grep online
Jun 29 12:45:46 vesnin sensor_manager2.py[827]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Jun 29 12:45:46 vesnin sensor_manager2.py[827]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Jun 29 12:45:46 vesnin sensor_manager2.py[827]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Jun 29 12:45:46 vesnin sensor_manager2.py[827]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
```
?

If yes, then see https://github.com/openbmc/openbmc/issues/1852 and https://github.com/openbmc/openbmc/issues/1870

Also please tell what are the results of

```
cat /sys/class/hwmon/hwmon0/device/modalias
cat /sys/class/hwmon/hwmon1/device/modalias
cat /sys/class/hwmon/hwmon2/device/modalias
```
- probably the results would be different for kernel 4.7 and kernel 4.10.

sdemirhan, 2017-06-29:
```
ls  -l /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/
total 0
-r--r--r--    1 root     root             4 Jun 29 13:40 #address-cells
-r--r--r--    1 root     root             4 Jun 29 13:40 #size-cells
-r--r--r--    1 root     root            11 Jun 29 13:40 compatible
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@100
drwxr-xr-x    3 root     root             0 Jun 29 13:40 i2c-bus@140
drwxr-xr-x    3 root     root             0 Jun 29 13:40 i2c-bus@180
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@1c0
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@300
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@340
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@380
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@3c0
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@40
drwxr-xr-x    3 root     root             0 Jun 29 13:40 i2c-bus@400
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@440
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@480
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@80
drwxr-xr-x    2 root     root             0 Jun 29 13:40 i2c-bus@c0
drwxr-xr-x    2 root     root             0 Jun 29 13:40 interrupt-controller@0
-r--r--r--    1 root     root             4 Jun 29 13:40 name
-r--r--r--    1 root     root            12 Jun 29 13:40 ranges

```
```
ls /sys/class/
bdi          hwmon        input        mem          net          ubi
block        i2c-adapter  leds         misc         rtc          watchdog
gpio         i2c-dev      mdio_bus     mtd          tty
```

```
cat /sys/class/hwmon/*/name 
ds3231
```
```
journalctl |grep online
Jun 29 10:53:55 garrison sensor_manager2.py[755]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Jun 29 10:53:55 garrison sensor_manager2.py[755]: sh: /sys/class/i2c-adapter/i2c-3/3-0050/online: No such file or directory
Jun 29 11:01:58 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:01:58 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:01:58 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:01:58 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:08:28 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:08:28 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:18:44 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:18:44 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:18:44 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:18:44 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:20:21 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:20:21 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:49:55 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:49:55 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:49:55 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:49:55 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:51:00 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:51:00 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
Jun 29 11:55:53 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/4-0050/online: Permission denied
Jun 29 11:55:53 garrison sensor_manager2.py[755]: sh: /sys/bus/i2c/devices/5-0050/online: Permission denied
```

```
cat /sys/class/hwmon/hwmon0/device/modalias
i2c:ds3231
```

there is only hwmon0.

Thank you...

mine260309, 2017-06-30:
@yadron You can use `/list` to list all the item under a PATH, or `/enumerate` to show all the properties of each item.
E.g.
```
curl -b cjar -k https://<bmc>/xyz/openbmc_project/sensors/list
curl -b cjar -k https://<bmc>/xyz/openbmc_project/sensors/enumerate
```

sdemirhan, 2017-06-30:
@mine260309 i use /list to list all the item under sensors but there is nothing.
```
curl -b cjar -k https://10.42.0.187/xyz/openbmc_project/sensors/list
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/sensors"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}
```

```
curl -b cjar -k https://10.42.0.187/xyz/openbmc_project/sensors/enumerate
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/sensors"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}
```

```
@ curl -b cjar -k https://10.42.0.187/xyz/openbmc_project/list
{
  "data": [
    "/xyz/openbmc_project/logging", 
    "/xyz/openbmc_project/led/groups/fan_fault", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm30", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core2", 
    "/xyz/openbmc_project/state/bmc0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core7", 
    "/xyz/openbmc_project/software", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core6", 
    "/xyz/openbmc_project/dump", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm6", 
    "/xyz/openbmc_project/inventory/system", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm7", 
    "/xyz/openbmc_project/led/groups/fan_identify", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1", 
    "/xyz/openbmc_project/led/groups/power_on", 
    "/xyz/openbmc_project/inventory", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard", 
    "/xyz/openbmc_project/software/18867d68", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core3", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm31", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core1", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm8", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core5", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core4", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm4", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm5", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core9", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core8", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm1", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm2", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm3", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core8", 
    "/xyz/openbmc_project/led/groups/bmc_booted", 
    "/xyz/openbmc_project/led/groups/enclosure_identify", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/membuf1", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/membuf0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/membuf3", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/membuf2", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/membuf5", 
[... 22 lines cut ...]
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm21", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm20", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm27", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm26", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm25", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm24", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm29", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm28", 
    "/xyz/openbmc_project/object_mapper", 
    "/xyz/openbmc_project/logging/internal/manager", 
    "/xyz/openbmc_project/led/groups", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core11", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core10", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core6", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core7", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core4", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core5", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core2", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core3", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core0", 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core1"
  ], 
  "message": "200 OK", 
  "status": "ok"
```

gtmills, 2018-05-23:
@sdemirhan Still a problem? If not, can you close?

gtmills, 2018-06-29:
Garrison was removed. Closing this issue.

closed 2018-06-29 by gtmills