openbmc/openbmc #1799: Partition failed to recover on checkstop

cdeadmin, 2017-06-16:


vishwabmc, 2017-06-16:
I updated CQ but not seeing it here.. 
```
=== State: Open by: vsubbann on 16 June 2017 07:01:32 ====

The Event data was not matching with the key code and value and the next read did not return any data so thats why GPIO Monitor keeps running.

Not sure if its something related to gpio-keys.

EV_READ_SUCCESS, EV_SYN AND REPORT

Brad:

    while (rc >= 0)
    {
        rc = libevdev_next_event(device.ge=== State: Open by: vsubbann on 16 June 2017 07:01:32 ====

The Event data was not matching with the key code and value and the next read did not return any data so thats why GPIO Monitor keeps running.

Not sure if its something related to gpio-keys.

EV_READ_SUCCESS, EV_SYN AND REPORT

Brad:

    while (rc >= 0)
    {
        rc = libevdev_next_event(device.get(),
                                 LIBEVDEV_READ_FLAG_NORMAL, &ev);

        if (rc == LIBEVDEV_READ_STATUS_SUCCESS)
        {

           if (ev.type == EV_SYN && ev.code == SYN_REPORT)
            {
                log<level::ERR>("VISHWA - EV_READ_SUCCESS, EV_SYN AND REPORT");
                continue;
            }

That returned SUCCESS on Iter:[0], but the ev.code and ev.value were [ EV_SYN ] and [ EV_REPORT ]

But when it went back and did read_next)event(), it did return < 0 . Looks like there were no more events ?.

===============

Is it something related to the key bindings ? t(),
                                 LIBEVDEV_READ_FLAG_NORMAL, &ev);

        if (rc == LIBEVDEV_READ_STATUS_SUCCESS)
        {

           if (ev.type == EV_SYN && ev.code == SYN_REPORT)
            {
                log<level::ERR>("VISHWA - EV_READ_SUCCESS, EV_SYN AND REPORT");
                continue;
            }

That returned SUCCESS on Iter:[0], but the ev.code and ev.value were [ EV_SYN ] and [ EV_REPORT ]

But when it went back and did read_next)event(), it did return < 0 . Looks like there were no more events ?.

===============

Is it something related to the key bindings ? ```

rfrandse, 2017-06-20:
<https://gerrit.openbmc-project.xyz/4862> Fix incorrect initialization of evdev
Fixes  #1799 Partition failed to recover on checkstop

closed 2017-06-20 by williamspatrick