openbmc/openbmc #1802: Detect OCC errors and notify userspace

geissonator, 2017-06-16:
Per the "OCC_P9_FW_Interface" specification, the BMC needs to monitor for these failures and take corrective action.

```
The BMC must send a request to reset the OCCs when it detects one of the following and the “OCC Active” sensor is TRUE with no checkstop present:
- Communication failure to an OCC defined in BMC-OCC Communication Failure section.
- Number of bits set in “OCCs Present” from master poll response does not match the number of OCCs the BMC is communicating with.
- Current OCC State byte in poll response is “Safe” for one minute and the “OCC Active” sensor is TRUE.
```

This story will track the driver work to detect these failures and provide a sysfs entry indicate the error that userspace can monitor.

eddiejames, 2017-06-16:
- [ ] Add non-hwmon sysfs attribute for occ error state
- [ ] Set the error state according to logic detailed above

bradbishop, 2017-07-13:
This has been merged.

closed 2017-07-13 by bradbishop