openbmc/openbmc #1803: Latest IR35221 driver causing intermittent boot issues

geissonator, 2017-06-16:
When a system hits this, it hits it consistently.  Some systems don't appear to hit it at all though.

Here's the error the host puts out and halts on:
```
 23.91755|ISTEP  8. 1 - host_slave_sbe_config
 24.44705|ISTEP  8. 2 - host_setup_sbe
 24.49958|ISTEP  8. 3 - host_cbs_start
 27.40536|================================================
 27.40537|Error reported by hwpf (0x0900) PLID 0x90FF00F6
 27.40538|  HW Procedure generated error. See User Data.
 27.40538|  ModuleId   0x01 MOD_FAPI2_RC_TO_ERRL
 27.40539|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
 27.40540|  UserData1  RC value from HWP : 0x000000000057cc3a
 27.40541|  UserData2  <unused> : 0x0000000000000000
 27.40542|------------------------------------------------
 27.48225|  HwpReturnCode              : RC_VDN_PGOOD_NOT_SET
 27.48225|  HWP Error description      : VDN_PGOOD not set to 1
 27.48226|------------------------------------------------
```

If we modify the new ir35221 scripts to just unbind the driver and leave it that way, things work ok.

Here's a dump of the voltages in the fail state:
```
[witherspoon]# /usr/bin/ir35221-unbind-bind.sh unbind
[witherspoon]# vrm-control.sh 
rail    set    read    current
-------    ------- ------- -------
vdda    1.000V    0.918V    7.000A
vcsa    1.031V    1.035V    0.000A
vdna    0.898V    0.801V    18.500A
vioa    1.051V    1.062V    5.125A
vddra    1.211V    1.211V    4.000A
vppa    2.512V    2.520V    0.000A
vddb    1.000V    0.918V    2.500A
vcsb    1.031V    1.031V    0.000A
vdnb    0.898V    0.801V    0.750A
viob    1.051V    1.062V    2.312A
vddrb    1.211V    1.211V    3.500A
vppb    2.512V    2.527V    0.000A
```

This driver came in last night's merge of the latest 4.10 changes
```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-5"
VERSION_ID="v1.99.7-5-g1afb614"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-5"
BUILD_ID="v1.99.7"
```

geissonator, 2017-06-16:
Potential fix is here - https://gerrit.openbmc-project.xyz/#/c/4851/

Not sure how it could be related to the IR35221 driver though, may just be we hit a new timing issue with the 4.10 + new IR35221 service changes.

rfrandse, 2017-06-16:
<https://gerrit.openbmc-project.xyz/4851> Enforce PAGE settings prior to vrm adjusts
Resolves  #1803 Latest IR35221 driver causing intermittent boot issues

closed 2017-06-19 by williamspatrick

PavamanSubramaniyam, 2017-07-03:
I am getting similar issue on P9 Zaius Hardware with host boot messages.

```
 69.30091|================================================
 69.30092|Error reported by hwpf (0x0900) PLID 0x9000000B
 69.30092|  HW Procedure generated error. See User Data.
 69.30093|  ModuleId   0x01 MOD_FAPI2_RC_TO_ERRL
 69.30094|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
 69.30095|  UserData1  RC value from HWP : 0x00000000005864fd
 69.30095|  UserData2  <unused> : 0x0000000000000000
 69.30096|------------------------------------------------
 69.30097|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30099|  HWP Error description      : I/O Xbus WDERF Training fail/timeout was reported.
 69.30100|------------------------------------------------
 69.30100|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30101|  FFDC                       : GROUP
 69.30102|    01
 69.30102|------------------------------------------------
 69.30103|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30104|  FFDC                       : TIMEOUT
 69.30104|    00
 69.30105|------------------------------------------------
 69.30106|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30107|  FFDC                       : M_WDERF_START
 69.30108|    00000100 5AFAF700
 69.30108|------------------------------------------------
 69.30109|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30110|  FFDC                       : M_WDERF_DONE
 69.30111|    00000000 000742F8
 69.30112|------------------------------------------------
 69.30112|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30113|  FFDC                       : M_WDERF_FAILED
 69.30114|    00000000 00191B08
 69.30115|------------------------------------------------
 69.30115|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30116|  FFDC                       : M_LANE_BAD_0_15
 69.30117|    00000000 00000000
 69.30118|------------------------------------------------
 69.30119|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30119|  FFDC                       : M_LANE_BAD_16_23
 69.30120|    00000000 0002336C
 69.30121|------------------------------------------------
 69.30122|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30123|  FFDC                       : M_LANE_DISABLED_VEC_0_15
 69.30124|    00000000 0004FD98
 69.30125|------------------------------------------------
 69.30125|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30126|  FFDC                       : M_LANE_DISABLED_VEC_16_23
 69.30127|    00000100 5AFAF700
 69.30128|------------------------------------------------
 69.30129|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30130|  FFDC                       : M_MAIN_INIT_STATE
 69.30131|    00000000 00000000
 69.30132|------------------------------------------------
 69.30133|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30133|  FFDC                       : M_WIRETEST_WTM_STATE
 69.30134|    00000000 00000000
 69.30135|------------------------------------------------
 69.30136|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30137|  FFDC                       : M_WIRETEST_WTR_STATE
 69.30138|    00000000 00000000
 69.30139|------------------------------------------------
 69.30140|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30141|  FFDC                       : M_WIRETEST_WTL_SM_STATUS
 69.30142|    00000000 00000000
 69.30142|------------------------------------------------
 69.30143|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 69.30144|  FFDC                       : M_WTR_BAD_LANE_COUNT
 69.30145|    00000000 00000000
 69.30146|------------------------------------------------
[... 175 lines cut ...]
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-74"
VERSION_ID="v1.99.7-74-gec3c51e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-74"
BUILD_ID="v1.99.7"

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x02820000..0x02821000 !
[                                                  ] 0%open-power-zaius-v1.17-111-gcc5a4f6-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-f576402
        linux-4.11.6-openpower1-p487dbeb
        petitboot-v1.4.3-p2462045
        machine-xml-5006472
        occ-7adb49a
        hostboot-binaries-0e2ad15
        capp-ucode-9c73e9f
        sbe-02021c6
[==================================================] 100%
```

Does the fixes pushed are available in the above Open BMC Firmware images.

geissonator, 2017-07-03:
I'm not seeing RC_VDN_PGOOD_NOT_SET in your error data @PavamanSubramaniyam?  I think you probably need a new issue.