openbmc/openbmc #1804: CORE and DIMM Present value as False

sdemirhan, 2017-06-19:
Based on the discussion in issue #428
I use garrison platform and i see all core and DIMM 'present is false. Do you have any recommends why i see like that?
I checked that ipmi works properly. 

root@garrison:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

root@ddyhpc-HP-ZBook-15-G3:/home/ddy-hpc# curl -b cjar -k https://10.42.0.40/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/bios": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/motherboard": {
      "fault": "False", 
      "fru_type": "MAIN_PLANAR", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/motherboard/apss": {
      "fault": "False", 
[... 488 lines cut ...]
    }, 
    "/org/openbmc/inventory/system/chassis/motherboard/todclock": {
      "fault": "False", 
      "fru_type": "MAIN_PLANAR", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/misc": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/systemevent": {
      "fault": "False", 
      "fru_type": "SYSTEM_EVENT", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }
  }, 
  "message": "200 OK", 
  "status": "ok"

williamspatrick, 2017-06-19:
All of the inventory should be in /xyz/openbmc_project/inventory not /org anymore.  It also looks like you are running a pretty old version of code.  Can you upgrade to v1.99.7?  You are currently over 200 commits back:

```
$ git rev-list v1.99.7 ^c133ec1 | wc -l
202
```

sdemirhan, 2017-06-19:
Thank you i upgrade openbmc to  v1.99.7.
-----
root@garrison:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-10"
VERSION_ID="v1.99.7-10-g521a594"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-10"
BUILD_ID="v1.99.7"
------

But when i want to read temperature sensors i did not access the repository. I read temperature sensors using restful after host booted.

curl -b cjar -k https://10.42.0.188/xyz/openbmc_project/Sensors/temperature/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/Sensors/temperature"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}

curl -b cjar -k https://10.42.0.188/xyz/openbmc_project/Sensors/temperature/ambient
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/Sensors/temperature/ambient"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}

Thank you for your time.

yadron, 2017-06-20:
I use latest "v1.99.7-5" at Barreleye, but still info regarding COREs and DIMMs and FANs and Motherboard and BMC-itself and etc presence not available:

```
curl b cjar -k https://172.17.27.28/org/openbmc/inventory/enumerate                                                   {
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/bios": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan4": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan5": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
...
    "/org/openbmc/inventory/system/chassis/motherboard": {
[... 504 lines cut ...]
    "/org/openbmc/inventory/system/chassis/sas_expander": {
      "fault": "False", 
      "fru_type": "DAUGHTER_CARD", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/misc": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/systemevent": {
      "fault": "False", 
      "fru_type": "SYSTEM_EVENT", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
```

sdemirhan, 2017-06-20:
I use garrison platform and i faced same result you got when i used /org/openbmc. But when i used /xyz/openbmc_project, i got the correct result. You can try  /xyz/openbmc_project/inventory to get info regarding COREs and DIMMs and FANs and Motherboard.

williamspatrick, 2017-06-21:
Yes.  /org/openbmc/inventory is deprecated in favor of /xyz path now.

closed 2017-06-21 by williamspatrick

yadron, 2017-06-21:
Yes, agree with that (that is closed), because just checked that "xyz" works better:

(and changed serial/part nums to xxx below):
```
curl -b cjar -k https://172.17.27.28/xyz/openbmc_project/inventory/enumerate
{
  "data": {
    "/xyz/openbmc_project/inventory/createme3": {
      "ExampleProperty1": "foo"
    }, 
    "/xyz/openbmc_project/inventory/system": {
      "BuildDate": "", 
      "Cached": 1, 
      "FieldReplaceable": 1, 
      "Manufacturer": "", 
      "Model": "1xxxG", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "", 
      "SerialNumber": "6xxx2"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/hdd_backplane": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "FOXCONN", 
      "Model": "", 
      "PartNumber": "1xxx-xxx-G", 
      "Present": 1, 
      "PrettyName": "BARRELEYE BPx15", 
      "SerialNumber": "1xxx-xxx-xxx2"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
      "BuildDate": "", 
      "Cached": 1, 
      "FieldReplaceable": 1, 
      "Manufacturer": "INGRASYS", 
      "Model": "", 
      "PartNumber": "1xxx-xxx-G", 
      "Present": 1, 
      "PrettyName": "BARRELEYE", 
      "SerialNumber": "6xxx2"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
      "BuildDate": "", 
      "Cached": 1, 
      "FieldReplaceable": 1, 
      "Functional": 1, 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "0xxx6", 
      "Present": 1, 
      "PrettyName": "PROCESSOR MODULE", 
      "SerialNumber": "Yxxx1", 
      "Version": "0xxx6"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core0": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core1": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core10": {
      "Functional": 1, 
      "Present": 1, 
      "PrettyName": ""
[... 158 lines cut ...]
      "Version": ""
    }, 
...
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm14": {
      "Functional": 0, 
      "Present": 0, 
      "PrettyName": ""
    }, 
...
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/membuf0": {
      "BuildDate": "", 
      "Cached": 1, 
      "FieldReplaceable": 0, 
      "Manufacturer": "INGRASYS", 
      "Model": "", 
      "PartNumber": "1...G", 
      "Present": 1, 
      "PrettyName": "BARRELEYE", 
      "SerialNumber": "6...2"
    }, 
...
  }, 
  "message": "200 OK", 
  "status": "ok"
```