openbmc/openbmc #1807: cores functional attribute is set to 1 but present is set to 0

jayeshmpatel, 2017-06-19:
"/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/core0": {
        "Functional": 1, 
        "Present": 0,                   
        "PrettyName": ""
}

I would think that if Functional: 1 then Present: 1 but not 0. I have seen it for some of the cores and not all. Others seems to have functional:1 and present:1

I have observed this on one of the Witherspoon with the following BMC code level and chassis/Host States.
 
"/xyz/openbmc_project/software/25bd815": {
                "Version": "ibm-v1.99.7-0-r0"
    }, 
       "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.On", 
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    }, 
    "/xyz/openbmc_project/state/host0": {
        "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running", 
        "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
    }, 

This might be due to the Host Firmware is not passing correct values to BMC. Please let me know if I should follow up with the Host Firmware.

gkeishin, 2017-06-20:
Is this the same with this https://github.com/openbmc/openbmc/issues/1357

jayeshmpatel, 2017-06-20:
Thanks George.. Yes, It looks the same hence closing this issue.

closed 2017-06-20 by jayeshmpatel