openbmc/openbmc #1808: IPL: DOA: phosphor-gpio-monitor core dump - failed to power on on zaius4 with 910.1726.20170616n

cdeadmin, 2017-06-20:


cbostic, 2017-06-20:
Data from CQ defect SW392619

==== State: Open by: andrewg on 19 June 2017 10:40:01 ==== 2

I think we need 2 issues here.  One for the core dump and one for the IPL fail. The IPL fail is due to this servie failing:

Jun 16 14:09:55 zaius kernel: fsi-master-gpio gpio-fsi: failed to get clock gpio
Jun 16 14:09:55 zaius kernel: fsi-master-gpio: probe of gpio-fsi failed with error -2
Jun 16 14:09:55 zaius kernel[915]: [  450.510000] fsi-master-gpio gpio-fsi: failed to get clock gpio
Jun 16 14:09:54 zaius phosphor-ledmanager[907]: No service providers for physical LED
Jun 16 14:09:55 zaius phosphor-ledmanager[907]: Asserting LEDs
Jun 16 14:09:55 zaius kernel[915]: [  450.510000] fsi-master-gpio: probe of gpio-fsi failed with error -2
Jun 16 14:09:54 zaius systemd[1]: Started Assert power_on LED.
Jun 16 14:09:54 zaius systemd[1]: Reached target Power0 On.
Jun 16 14:09:55 zaius sh[1230]: /bin/sh: line 0: echo: write error: No such device
Jun 16 14:09:54 zaius systemd[1]: Reached target Power0 (On).
Jun 16 14:09:54 zaius systemd[1]: Starting Bind FSI Driver...
Jun 16 14:09:54 zaius systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Jun 16 14:09:54 zaius systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Jun 16 14:09:54 zaius systemd[1]: Starting Fix up Zaius host refclock...
Jun 16 14:09:54 zaius systemd[1]: Started Phosphor Fan Control Initialization.
Jun 16 14:09:56 zaius sh[1230]: /bin/sh: line 0: echo: write error: No such device
Jun 16 14:09:55 zaius systemd[1]: Started Fix up Zaius host refclock.
Jun 16 14:09:56 zaius systemd[1]: fsi-bind@0.service: Main process exited, code=exited, status=1/FAILURE
Jun 16 14:09:56 zaius systemd[1]: Failed to start Bind FSI Driver.
Jun 16 14:09:56 zaius systemd[1]: Dependency failed for Chassis0 (Power On).
Jun 16 14:09:56 zaius systemd[1]: Dependency failed for Start Host0.
Jun 16 14:09:56 zaius systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Jun 16 14:09:56 zaius systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Jun 16 14:09:56 zaius systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
Jun 16 14:09:56 zaius systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
Jun 16 14:09:56 zaius systemd[1]: fsi-bind@0.service: Unit entered failed state.
Jun 16 14:09:56 zaius systemd[1]: fsi-bind@0.service: Failed with result 'exit-code'.

There's a lot of kernel backtraces to look at, I think we have a separate issue for that.

==== State: Open by: andrewg on 19 June 2017 10:41:33 ==== 3

#=#=# 2017-06-19 10:41:30 (CDT) #=#=#
New OwnerInfo = [Bostic, Christopher l. (cbostic@us.ibm.com)]

Lets use this to track the FSI bind service failure.  Assigning to Chris.
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#

==== State: Open by: cbostic on 19 June 2017 13:04:25 ==== 4

So the bind failed because the GPIO FSI driver could not retrieve its CLOCK pin (C3) from the GPIO driver.

I see that C3 a.k.a GPIO #299 is being grabbed by some user space code.

root@zaius:/sys/class/gpio# ls -l
total 0
--w-------    1 root     root          4096 Jun 19 15:17 export
lrwxrwxrwx    1 root     root             0 Jun 19 15:01 gpio292 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio292
lrwxrwxrwx    1 root     root             0 Jun 19 15:17 gpio299 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio299
lrwxrwxrwx    1 root     root             0 Jun 19 15:17 gpio300 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio300
lrwxrwxrwx    1 root     root             0 Jun 19 15:01 gpio305 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio305
lrwxrwxrwx    1 root     root             0 Jun 19 15:01 gpio306 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio306
lrwxrwxrwx    1 root     root             0 Jun 19 15:01 gpio347 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio347
lrwxrwxrwx    1 root     root             0 Jun 19 15:01 gpio417 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio417
lrwxrwxrwx    1 root     root             0 Jun 19 15:01 gpio488 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpiochip0/gpio/gpio488
lrwxrwxrwx    1 root     root             0 Jun 19 15:00 gpiochip280 -> ../../devices/platform/ahb/ahb:apb/1e780000.gpio/gpio/gpiochip280
--w-------    1 root     root          4096 Jun 19 15:00 unexport
root@zaius:/sys/class/gpio# gpioutil -l c3
299

==== State: Open by: cbostic on 19 June 2017 21:45:34 ==== 5

This commit in the dev-4.10 Linux kernel seems to cause the problems seen.   The commit fixes issues on the EVT board but there are many installations that aren't using this yet.

[... 16 lines cut ...]
--- a/arch/arm/boot/dts/aspeed-bmc-opp-zaius.dts
+++ b/arch/arm/boot/dts/aspeed-bmc-opp-zaius.dts
@@ -80,8 +80,12 @@

                status = "okay";

-               clock-gpios = <&gpio ASPEED_GPIO(C, 3) GPIO_ACTIVE_HIGH>;
-               data-gpios = <&gpio ASPEED_GPIO(C, 2) GPIO_ACTIVE_HIGH>;
+               /*
+                * These 2 GPIOs are set per board revision in aspeed.c.
+                * These are the DVT and beyond values:
+                * clock-gpios = <&gpio ASPEED_GPIO(G, 0) GPIO_ACTIVE_HIGH>;
+                * data-gpios = <&gpio ASPEED_GPIO(G, 1) GPIO_ACTIVE_HIGH>;
+                */
                trans-gpios = <&gpio ASPEED_GPIO(O, 6) GPIO_ACTIVE_HIGH>;
                enable-gpios = <&gpio ASPEED_GPIO(D, 0) GPIO_ACTIVE_HIGH>;
                mux-gpios = <&gpio ASPEED_GPIO(P, 6) GPIO_ACTIVE_HIGH>;

Sending to Robert to comment.

==== State: Open by: cbostic on 19 June 2017 21:46:07 ==== 6

Reverting this patch allows zaius1 to IPL properly.

==== State: Open by: cbostic on 19 June 2017 21:49:04 ==== 7

geissonator, 2017-06-20:
@GHF - Can't seem to find an ID for Rob.  He put a patch into 4.10 that appears to have broken master from booting on Zaius.  Could you talk with him on this?

GHF, 2017-06-20:
I just took a look. It looks like the dynamic mapping he added for FSI GPIO assignments is against the device called `fsi-master`, because he wrote the patch against the 4.7 tree.
https://github.com/openbmc/linux/blob/fb92f38f80699b2089c3467b1344ac107f797d3c/arch/arm/boot/dts/aspeed-bmc-opp-zaius.dts#L78

In the 4.10 OpenBMC kernel, the FSI master device is called `gpio-fsi`.
https://github.com/openbmc/linux/blob/f77d9dc09ebd173f9b81abf4f31f12fe853f924c/arch/arm/boot/dts/aspeed-bmc-opp-zaius.dts#L78
I'll ask Rob to send a new version of the change, this time against the 4.10 kernel.

geissonator, 2017-07-05:
@GHF - Can we close this one out?  I think I saw something come across on the mailing list for it?

GHF, 2017-07-07:
@geissonator Sure, please close.

closed 2017-07-07 by williamspatrick