openbmc/openbmc #1810: IPMI fails with "Unspecified Error" during BMC boot

stewart-ibm, 2017-06-20:
While BMC is booting:
```
$ ipmitool -I lanplus -H rom1-bmc -P 0penBmc chassis power status
Get Chassis Power Status failed: Unspecified error
$ ipmitool -I lanplus -H rom1-bmc -P 0penBmc chassis power status
Chassis Power is off
```

stewart-ibm, 2017-06-20:
```
root@romulus:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-34"
VERSION_ID="v1.99.5-34-g2ce4c6a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-34"
BUILD_ID="v1.99.5"
```

williamspatrick, 2017-06-20:
Ok?  Is there something better it should fail with when we haven't discovered the state yet?

gtmills, 2018-05-31:
@stewart-ibm Still a problem or can this be closed?

stewart-ibm, 2018-06-01:
I haven't tried, we switched to using exclusively the REST interface while IPMI wasn't suitable and haven't gone back.

closed 2019-04-27 by stale[bot]