openbmc/openbmc #1811: IPMI SOL output 15 minutes behind ssh console

stewart-ibm, 2017-06-20:
```
root@romulus:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-34"
VERSION_ID="v1.99.5-34-g2ce4c6a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-34"
BUILD_ID="v1.99.5"
```

SSH to port 2200 console is working as expected. If you connect to IPMI SoL though, it's still displaying Hostboot output 19 seconds into boot after the host has been up (and functioning, and interacting with it) for over 13 minutes.

stewart-ibm, 2017-06-20:
Also reproduces on
```root@rom1-bmc:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-3"
VERSION_ID="v1.99.7-3-gd0c8db0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-3"
BUILD_ID="v1.99.7"
```

yadron, 2017-06-20:
@stewart-ibm , it could be software-hardware problems with interrupts or inside CPLD. What is the platform that you are using? Is it P8 or P9 and what is the hardware?

I'm using "v1.99.7-5" that works well at Barreleye both if direct SOL from BMC to Host or if using SSH SOL.   From the other point of view, I've seen similar "frozen during host-booting" (or wait for 10 secs after each button press) that was because of bad consts inside CPLD (so recompiling CPLD with upgrading timing const regarding interrupts can help then).

Also I can recommend upgrading PNOR to latest version, because both PNOR or CPLD content could be the reason of what you see.

Also, if any hardware bug with interrupts (for example, check if output of this cmd at host Linux expected:
`cat /proc/interrupts`
- see https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/6/html/Performance_Tuning_Guide/s-cpu-irq.html
), then it could be the reason too.
Let me explain that SOL from BMC to Host works using interrupts too (or uses the same hardware that is being used for interrupts), so if interrupts overflow happens with any other non-related to SOL interrupt or other things with interrupts (for example interrupts frozen for 15 mins like you've seen or interrupts incorrectly configured inside CPLD), then SOL frozen too.

stewart-ibm, 2017-06-21:
Alex <notifications@github.com> writes:
> @stewart-ibm , it could be software-hardware problems with interrupts or inside CPLD. What is the platform that you are using? Is it P8 or P9 and what is the hardware?
>
> I'm using "v1.99.7-5" that [...]

IIRC this was on Romulus or Witherspoon (both?)

The ssh connection (port 2200) for the console was keeping up fine and wasn't a
problem at all, it was just the SoL connection.

-- 
Stewart Smith
OPAL Architect, IBM.

tomjoseph83, 2017-06-21:
@stewart-ibm Did you get to the petit-boot prompt eventually? I have observed delay in displaying console data in IPMI SOL compared to obmc-console-client over SSH. This was observed when hostboot/SBE detailed tracing was enabled. One IPMI SOL packet is limited to 255 bytes.

stewart-ibm, 2017-06-23:
@tomjoseph83 to be honest, I didn't wait that long.

But yes, I was running builds with verbose output from hostboot, so there's a *lot* of output on boot.

What was interesting though was that I only attached IPMI SoL after boot, and then I saw messages from much earlier, I would have expected it not to buffer things while there was no SOL connection.

rfrandse, 2017-08-01:
<https://gerrit.openbmc-project.xyz/5693> Cleanup the console buffer on stopping the SOL payload.
Resolves  #1811 IPMI SOL output 15 minutes behind ssh console

closed 2017-08-02 by williamspatrick