openbmc/openbmc #1812: temporary Insufficient privilege level while power off/on cycle

stewart-ibm, 2017-06-20:
```
$ ipmitool -I lanplus -H rom1-bmc -P 0penBmc power off
Chassis Power Control: Down/Off
$ ipmitool -I lanplus -H rom1-bmc -P 0penBmc power on
> Error: no response from RAKP 1 message
Set Session Privilege Level to ADMINISTRATOR failed: Insufficient privilege level
Error: Unable to establish IPMI v2 / RMCP+ session
$ ipmitool -I lanplus -H rom1-bmc -P 0penBmc power on
Chassis Power Control: Up/On
```

```
root@rom1-bmc:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-3"
VERSION_ID="v1.99.7-3-gd0c8db0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-3"
BUILD_ID="v1.99.7"
```

tomjoseph83, 2017-08-05:
@stewart-ibm

> Error: no response from RAKP 1 message
The above log message means session setup failed. In which case no IPMI commands can be sent.
So the session privilege command would not succeed  and the following log is expected.
Set Session Privilege Level to ADMINISTRATOR failed: Insufficient privilege level

We do not have the IPMI user account implemented and the sessions are given ADMIN privilege level.
So insufficient privilege level error has nothing to do with the implementation of the command.

There is no action for me to take on this issue.

geissonator, 2017-08-05:
But what caused this error Tom?
```
$ ipmitool -I lanplus -H rom1-bmc -P 0penBmc power on
> Error: no response from RAKP 1 message
Set Session Privilege Level to ADMINISTRATOR failed: Insufficient privilege level
```

Do we need the journal to debug? I think we have some other bugs on intermittent IPMI command fails?

tomjoseph83, 2017-08-07:
@geissonator This is a failure on the ipmitool side. Probably this is a bug on the ipmitool that it issues session privilege command after session setup failure. Detailed ipmitool traces in such scenario would help.

Journal wouldn't do any help in this case. I have not seen any other bug that reports intermittent session setup failures. Please point me in case of any.

geissonator, 2017-08-07:
I was thinking of #2064 but I guess that's not related here?

tomjoseph83, 2017-08-08:
#2065 is not related to this issue.

rfrandse, 2017-09-01:
<https://gerrit.openbmc-project.xyz/6359> Cache system guid in netipmid
Resolves  #1812 temporary Insufficient privilege level while power off/on cycle

closed 2017-09-11 by williamspatrick