openbmc/openbmc #1813: slpd[901]: SLP unable to find the service=service:directory-agent

PavamanSubramaniyam, 2017-06-20:
On Zaius Hardware we get the following messages in journalctl.

```
Apr 20 17:48:37 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:48:38 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:48:39 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:48:40 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:48:41 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:20 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:22 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:38 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:40 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:40 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:42 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:47 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:49:49 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:50:26 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:50:28 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:51:12 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:51:12 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:51:14 zaius slpd[901]: SLP unable to find the service=service:directory-agent
Apr 20 17:51:14 zaius slpd[901]: SLP unable to find the service=service:directory-agent
```

PavamanSubramaniyam, 2017-06-20:
The machine is installed with the following OpenBMC and PNOR Firmware images.

```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-13"
VERSION_ID="v1.99.7-13-gb0261ea"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-13"
BUILD_ID="v1.99.7"

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-zaius-v1.17-65-g4e80220
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-87-ge835c2fb7ce1
        hostboot-44d178b
        linux-4.11.2-openpower1-pef1c0df
        petitboot-v1.4.2-p2462045
        machine-xml-5006472
        occ-1dc97a6
        hostboot-binaries-24995fc
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%
```

PavamanSubramaniyam, 2017-06-20:
[journalctl.txt](https://github.com/openbmc/openbmc/files/1087244/journalctl.txt)
[dmesg.txt](https://github.com/openbmc/openbmc/files/1087247/dmesg.txt)

ratagupt, 2017-07-07:
slp-lite doesn't support "service:directory agent"

williamspatrick, 2017-07-07:
For some reason we get a flood of these in the journal though.  We should do something to reduce the frequency of journal entries if someone is making the same incorrect query over and over.

It seems like there is something in this lab environment that is polling all machines for their SLP data.

GHF, 2017-07-07:
I'm not even sure what slpd is...

williamspatrick, 2017-07-07:
@GHF - https://github.com/openbmc/slpd-lite

GHF, 2017-07-07:
@williamspatrick Follow up question, what is SLP? Is it this? https://www-01.ibm.com/software/webservers/hostondemand/library/infocentergafinal/hod/en/help/slp.html

I take it that it probably won't impact a non-IBM environment too much?

gtmills, 2018-08-14:
@PavamanSubramaniyam @GHF Can this be closed?

PavamanSubramaniyam, 2018-08-28:
Even with the latest Open BMC upstream build for Zaius, we still see these messages in journalctl logs:

```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.3-192"
VERSION_ID="v2.3-192-g35497dc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.3-192"
BUILD_ID="v2.3"

root@zaius:~# journalctl -f
-- Logs begin at Tue 2018-08-28 07:00:10 UTC. --

Aug 28 07:04:08 zaius slpd[984]: [24B blob data]
Aug 28 07:04:08 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:05:21 zaius slpd[984]: [24B blob data]
Aug 28 07:05:21 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:05:23 zaius slpd[984]: [24B blob data]
Aug 28 07:05:23 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:06:00 zaius slpd[984]: [24B blob data]
Aug 28 07:06:00 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:06:02 zaius slpd[984]: [24B blob data]
Aug 28 07:06:02 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:09:36 zaius slpd[984]: [24B blob data]
Aug 28 07:09:36 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:09:38 zaius slpd[984]: [24B blob data]
Aug 28 07:09:38 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:09:46 zaius slpd[984]: [24B blob data]
Aug 28 07:09:46 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:09:47 zaius slpd[984]: [24B blob data]
Aug 28 07:09:47 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:09:48 zaius slpd[984]: [24B blob data]
Aug 28 07:09:48 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:09:49 zaius slpd[984]: [24B blob data]
Aug 28 07:09:49 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:10:00 zaius slpd[984]: [24B blob data]
Aug 28 07:10:00 zaius slpd[984]: SLP unable to find the service=service:directory-agent
Aug 28 07:10:02 zaius slpd[984]: [24B blob data]
Aug 28 07:10:02 zaius slpd[984]: SLP unable to find the service=service:directory-agent
```

closed 2019-04-26 by stale[bot]