openbmc/openbmc #1814: uptime: couldn't get boot time: No such file or directory

stewart-ibm, 2017-06-20:
```
root@romulus:~# uptime
uptime: couldn't get boot time: No such file or directory
root@romulus:~# which uptime
/usr/bin/uptime
root@romulus:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-3"
VERSION_ID="v1.99.7-3-gd0c8db0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-3"
BUILD_ID="v1.99.7"
root@romulus:~# cat /proc/uptime 
666.56 532.43
root@romulus:~# 
```

williamspatrick, 2017-06-20:
That version does not exist on github.  What is the code you are building / running with?

A witherspoon works with nearly-latest code:
```
root@witherspoon:~# uptime
 10:19:29 up 15:47,  load average: 0.35, 0.32, 0.34
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-3"
VERSION_ID="v1.99.7-3-g97c6018-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-3"
BUILD_ID="v1.99.7"
```
```
commit 97c6018e9108e8ace952dace80d7ad45ff5f936a (origin/master, origin/HEAD, refs/changes/13/4813/2)
Author: Deepak Kodihalli <dkodihal@in.ibm.com>
Date:   Thu Jun 15 02:08:42 2017 -0500

    React to boxelder eeprom path change
    
    The path of the BMC eeprom (on Witherspoon's boxelder card) has changed
    in the 4.10 kernel. This path serves as an input to a vpd parsing
    program. Change the path in that program's bitbake recipe.
    
    Change-Id: I7559a2d6fa89444fcea591956ad908dd296c592e
    Signed-off-by: Deepak Kodihalli <dkodihal@in.ibm.com>
```

stewart-ibm, 2017-06-21:
It was a build from Lei Yu  for Romulus that I got pointed to in the (internal) Romulus slack channel.

anoo1, 2017-06-26:
Milton mentions that strace could be used to check what uptime is trying to read.

gtmills, 2018-05-31:
@stewart-ibm Still a problem or can this be closed?

closed 2018-06-13 by anoo1