openbmc/openbmc #1818: Support to directly write vrm voltages to device driver

geissonator, 2017-06-20:
There's a need to write voltage overrides to the VRM's.  Currently we do this via bash scripts that run i2cset commands.  This will be ok for the initial GA, but long term we'd like the decvice drivers for the VRM's to support direct voltage writes.  That way we can get away from these i2cset commands.  Probably need 2 stories here, 1 to support this in the driver and 1 to use the new driver.

jkeusema, 2017-06-20:
@geissonator need to look at locking scheme for device driver as I outlined in SW390917 due to PAGE + register structure of these VRMs.

gtmills, 2018-07-18:
@geissonator Still needed?

geissonator, 2019-02-25:
Assigning to Derek to take a look (and maybe talk with Jordan). If needed we should probably work with @rfrandse to move over to the ibm github project.

closed 2019-10-27 by stale[bot]