openbmc/openbmc #1819: Sprint 17.06.28 - Committed

rfrandse, 2017-06-20:


dkodihal, 2017-06-27:
***Squad 1***

**Devender**
#1560 IPMI FRU commands : implementation
#1772 phosphor-logging : improve metadata provision

**Dhruv**
#1785 Persist host and chassis state requests

**Vishwa**
#1363 Request OCC reset when fail detected

**Jayanth**
#1795 BMC Dump todos

**Ratan**
Vacation Jun 29 - Jul 5

**Tom**

**Deepak**

bradbishop, 2017-06-27:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Gunnar:gtmills: 80%**
 **17.06.28:**

- **#1329 (power supply presence)**
Add a story from #1707 or #1708 here.

vacation: 7/17 - 7/21

-------------------------------------------------------------------------------------------------
**Matt S:spinler: 80%** 
**17.06.28:**

- #1342 (base Watchdog support - install and run busybox watchdog) 
      - 6/21 - ready for merge, but need #1636.

- #1724 (chip fault analysis)

-------------------------------------------------------------------------------------------------
**Matt B:msbarth: 80%**
**17.06.28:**

- **#1626 (calculate speed change request)**
   
- **#1625 (loop)**

- #1835 (delay on events parameter)

- #1839 (handle scale property)

vacation: 7/12 - 7/25

design topics:
  - fan overrides
-------------------------------------------------------------------------------------------------
**Brad:bradbishop**: 70%

**17.05.31:**
- **#1318 (gpio presence)**
- **#1300 (real error logs in presence)**

**17.06.14:**
- **#1186 (115C shutdown)**
- **#1188 (both fan rpms)**

-------------------------------------------------------------------------------------------------
**Chris:cbostic**: 75%

**17.06.14:**
- **#1636 (watchdog driver DT)**
     - 6/26 - will send joel a note

- #1783 (witherspoon gpio-keys)
      - 6/26 - have a solution?  working today

**17.06.28:**

vacation: half day 6/29

next sprint: penciled in for power supply driver?
------------------------------------------------------------------------------------------------
**Eddie:eddiejames**: 80%

**17.06.14:**
- **#1698 (fsi rescan for occ devices)**
    - 6/16 - working on it
    - 6/19 - what to do with this story?
    - 6/22 - Vishwa doing this?

- #1802 (occ errors)
     - 6/19 - just getting started
     - 6/20 - most logic finished, testing.
     - 6/21 - tested, things working, pushed to mailing list.

upstreaming:
    - 6/22 - gkh fixes for sbefifo, hwmon next
    - 6/23 - hwmon driver out to list, add i2c devices to device tree

next sprint: need something for next sprint.

-------------------------------------------------------------------------------------------------
**Brandon:bjwyman**: 50%

**17.06.14:**
 - #1738 (fan fault led device)
      - 6/16 - chip shows up, looking for DT examples
      - 6/19 - looking for help.
      - 6/20 - still looking for help.
      - 6/21 - netbooting might not be working?
      - 6/22 - looking at it.
      - 6/23 - some questions in group chat, possible issue with hypen vs underscore

vacation: out 6-27 to 7-4
-------------------------------------------------------------------------------------------------
**Dinesh:chinaridinesh**: 65%

**17.06.14:**
- **#1315 (Ensure uboot fw_env update script uses new inventory object)**
    - 6/16 - review comments to address
    - 6/19 - comments addressed (testing uuid implementation, hopefully done today)
    - 6/20 - uuid pushed up, ready for review
    - 6/21 - need a review still
    - 6/22 - need a review
    - 6/23 - need a review
    - 6/26 - need a review

- **#1555 (make real error logs in fan control)**
    - 6/16 - need one more plus one
    - 6/19 - ready for merge review.
    - 6/20 - ready for merge review.
    - 6/22 - ready for merge review.
    - 6/23 - ready for merge review.
    - 6/26 - needs a rebase.

- #1322 (pgood is lost on BMC reboot on witherspoon)
    - 6/21 - spent some time here.
    - 6/23 - still poking at it.
    - 6/26 - still working

next sprint: need a story

vacation: 7-10 (half day), 7-11

anoo1, 2017-06-29:
# Squad 3 -

**Adriana:anoo1 -**
**17.03.08**
#1297 eSEL Missing AdditionalData [ Witherspoon ] 
  * 6/29 - Per Patrick, more investigation is needed to propose a solution that would be acceptable upstream.

**17.06.14**
- #1651 BMC filesystem: Create UBI read-only and read-write partitions
  * 6/29 - Per Patrick, we want to remove the initramfs and mount the partitions from the kernel.

**17.06.28**
- #1758 BMC: Flag to specify the chips where partitions should be stored
- #1333 PNOR: Enable patching for lab/mfg
- #1551 mboxd: Use tmp location for pnor patches
---

**Andrew:geissonator -**

**17.06.28**
- #948 Customer Power Cap

- #1161 Witherspoon: System Failed to respond from 2nd iteration and serial console hangs (bug)
  * 6/29 - Talked with systemd developers, continuing with debug.

---

**Michael:mtritz -**

**17.04.05**
- #1351 
  * 6/29 - Just a makefile change to do and then re-submit to mailing list.

**17.06.28**
- #1576 Implement BMC factory reset
  * 6/29 - Created task list.

- #1653 Implement Field Mode

---

**Saqib:saqibkh -**

**17.06.14**
- #1553 PNOR: Implement Software RedundancyPriority.interface
  * 6/29 - Re-uploaded change based on Patrick's feedback, need re-review.

**17.06.28**
- #1755 BMC: Implement Software RedundancyPriority.interface

- #1650 Move contents from /var/lib/obmc to /var/persist

---

**Leo:lgon -**

**17.06.28**
- #1550 Implement Software Object.Delete for pnor updater and image manager
  * 6/29 - First commit up for review

- #1842 BMC: Create Manifest file and add it to the <system>.tar image

closed 2017-07-20 by rfrandse