openbmc/openbmc #1821: Provide throttling and overtemp information to external users

geissonator, 2017-06-20:
#1619 provides new hwmon entries for things like throttling due to overtemps and power limits.  When these get set, we need something to provide this information to the user (via event log or some other mechanism).   This issue is to track that work.

geissonator, 2017-08-03:
We'll use this story to add the new sensors to hwmon.  #1966 for notifying people of the events.

eddiejames, 2017-08-09:
# **STATUS**

08/09 - No progress yet due to working on pmbus core.

dkodihal, 2017-09-05:
This would need to use the event logging mechanism (#1856) to be able to log the event, once the overtemp info is known.

geissonator, 2017-09-05:
Eddie to look into using alarm feature for over temps.  
Reporting throttling?  debugfs?  Some non-standard mechanism.

rfrandse, 2017-10-04:
<https://gerrit.openbmc-project.xyz/7056> Add watches for throttling reported by the OCC
Resolves  #1821 Provide throttling and overtemp information to external users

closed 2017-12-12 by geissonator