openbmc/openbmc #1822: [Barreleye] org.openbmc.control.Host@0.service core-dump

swe12345, 2017-06-20:
```
Apr 28 13:27:38 barreleye control_host.exe[902]: **
Apr 28 13:27:38 barreleye control_host.exe[902]: ERROR:control_host_obj.c:218:on_boot: assertion failed (error == NULL): GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.watchdog.Host was not provided by any .service files (g-dbus-error-quark, 2)
Apr 28 13:27:38 barreleye phosphor-host-state-manager[904]: Received signal that host is running
```
```
Apr 28 13:27:38 barreleye systemd[1]: org.openbmc.control.Host@0.service: Main process exited, code=dumped, status=6/ABRT
Apr 28 13:27:38 barreleye systemd[1]: org.openbmc.control.Host@0.service: Unit entered failed state.
Apr 28 13:27:38 barreleye systemd[1]: org.openbmc.control.Host@0.service: Failed with result 'core-dump'.
```
Driver:
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-13"
VERSION_ID="v1.99.7-13-gb0261ea"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-13"
BUILD_ID="v1.99.7"
```
```
root@barreleye:~# uname -a
Linux barreleye 4.10.17-09041b7c3b65a7ec1063d4882b58fd6f00fc87b3 #1 Mon Jun 19 17:51:01 UTC 2017 armv5tejl GNU/Linux
````
[journald_v1.99.7-13-gb0261ea_host_control_core.txt](https://github.com/openbmc/openbmc/files/1089053/journald_v1.99.7-13-gb0261ea_host_control_core.txt)
```

geissonator, 2017-06-20:
New watchdog needs to be added to barreleye recipes @vishwabmc ?

vishwabmc, 2017-06-21:
@geissonator New watchdog is a common one under phosphor and is pulled by State Manager. So it does not need anything specific to Barreleye.

But I see this control_host_obj.c is referring to old Watchdog object. May be this particular program from skeleton is still being used to achieve fsi bitbang. WIll talk to Patrick and update.

rfrandse, 2017-06-21:
<https://gerrit.openbmc-project.xyz/4907> Remove references to old watchdog object
Fixes  #1822 [Barreleye] org.openbmc.control.Host@0.service core-dump

closed 2017-06-21 by williamspatrick