openbmc/openbmc #1824: OCC pass_through controller no longer shows up in the systemctl list

essawy, 2017-06-20:
flashing today's BMC master (witherspoon-20170620151634.all.tar) on a witherspoon, the OCC pass_through controllers were no longer showing up in the buctl list, even though the occ devices where showing under /dev:

```
root@witherspoon:~# systemctl restart org.open_power.OCC.PassThrough

root@witherspoon:~# ls /dev/occ*
/dev/occ1  /dev/occ2

root@witherspoon:~# busctl tree | grep -i OCC
Service org.open_power.OCC.PassThrough:
    └─/xyz/openbmc_project/occ
      └─/xyz/openbmc_project/occ/pass_through
        ├─/org/freedesktop/systemd1/unit/org_2eopen_5fpower_2eOCC_2ePassThrough_2eservice
        │ └─/org/openbmc/sensors/host/cpu0/OccStatus
        │ └─/org/openbmc/sensors/host/cpu1/OccStatus

root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-14"
VERSION_ID="v1.99.7-14-gfbe25fc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-14"
BUILD_ID="v1.99.7"
```

I flashed an older BMC image (witherspoon-20170613193050.all.tar) to verify whether the issue is a BMC image issue, then the two occ pass_through controllers showed up again in the busctl list:

```
root@witherspoon:~# ls /dev/oc*
/dev/occ1  /dev/occ2
root@witherspoon:~# systemctl restart org.open_power.OCC.PassThrough
root@witherspoon:~# busctl tree | grep -i OCC
Service org.open_power.OCC.PassThrough:
    └─/xyz/openbmc_project/occ
      └─/xyz/openbmc_project/occ/pass_through
        ├─/xyz/openbmc_project/occ/pass_through/occ0
        └─/xyz/openbmc_project/occ/pass_through/occ1
^C
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-130"
VERSION_ID="v1.99.6-130-g35cfa84-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-130"
BUILD_ID="v1.99.6"
```

dkodihal, 2017-06-21:
@essawy You don't need that workaround of restarting the OCC pass-through service now. The occ d-bus objects should get created in response to cpu objects getting created in the inventory. In other words you should see the occ objects once the box has booted to host runtime. Let me know if you don't see that happening.

vishwabmc, 2017-06-21:
@dkodihal You were going to make some updates about populating these dbus objects based on inventory ?. Is that in ?. If so, is this related to that ?

essawy, 2017-06-21:
@dkodihal @vishwabmc I neither saw the occ dbus objects using the busctl command (shown above) nor saw them in the pass_through enumerate curl command output:

```curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET   https://$bmc/xyz/openbmc_project/occ/pass_through/enumerate 
curl: (6) Couldn't resolve host 'application'
{
  "data": {}, 
  "message": "200 OK", 
  "status": "ok"```

dkodihal, 2017-06-22:
@essawy I got on to a Witherspoon today (it had host running already). This is what I see :

root@witherspoon:~# busctl tree org.open_power.OCC.PassThrough

/xyz
  /xyz/openbmc_project
    /xyz/openbmc_project/occ
      /xyz/openbmc_project/occ/pass_through
        /xyz/openbmc_project/occ/pass_through/occ0
        /xyz/openbmc_project/occ/pass_through/occ1

Although due to a known mapper bug, they won't show up in a REST query : #1767.

essawy, 2017-06-22:
Also, restarting the service takes the controllers out of the busctl list

```
root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@witherspoon:~#  busctl tree org.open_power.OCC.PassThrough | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/occ
      └─/xyz/openbmc_project/occ/pass_through
root@witherspoon:~# obmcutil poweron
root@witherspoon:~#  busctl tree org.open_power.OCC.PassThrough | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/occ
      └─/xyz/openbmc_project/occ/pass_through
        ├─/xyz/openbmc_project/occ/pass_through/occ0
        └─/xyz/openbmc_project/occ/pass_through/occ1
root@witherspoon:~# systemctl restart org.open_power.OCC.PassThrough
root@witherspoon:~#  busctl tree org.open_power.OCC.PassThrough | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/occ
      └─/xyz/openbmc_project/occ/pass_through
```

I understand that in this specific case, I don't need to restart the service. But in general, shouldn't the dbus occ objects remain there after service restart?

dkodihal, 2017-06-22:
> I understand that in this specific case, I don't need to restart the service. But in general, shouldn't the dbus occ objects remain there after service restart?

Yes, they should be recreated. I have to push a commit to get that working.

dkodihal, 2017-06-24:
I'll use this defect to fix the problem mentioned in the previous comment.

dkodihal, 2017-07-24:
# **STATUS**

24 July : Planning to start this tomorrow. Code should be up for review by 26th.

rfrandse, 2017-07-26:
<https://gerrit.openbmc-project.xyz/5544> Reconstruct OCC objects on app restart
Resolves  #1824 OCC pass_through controller no longer shows up in the busctl list

closed 2017-08-02 by williamspatrick