openbmc/openbmc #1825: BMP280 HWMON process missing

bjwyman, 2017-06-20:
After the updates to find the correct instance path for iio hwmon drivers, there is still some kind of problem with the process not running when a system is booted up.
Using the command line to unbind/bind the device also does not get the device to show up and stick around.
It is not clear if this also affects the dps310 device, as that driver was not available to check for the process.

bjwyman, 2017-06-21:
One piece of the puzzle might be a missing rules file. There is a 70-hwmon.rules, possibly need a 70-iio.rules as well.

rfrandse, 2017-06-22:
<https://gerrit.openbmc-project.xyz/4913> Add udev rule file for iio-hwmon type drivers
Resolves  #1825 BMP280 HWMON process missing

closed 2017-06-26 by williamspatrick