openbmc/openbmc #1826: irq 22: nobody cared

shenki, 2017-06-21:
On Romulus running 4.10.17-c694e47cbc1efc2bc181b2ac7137dfda04f974c2 v1.99.7-3-gd0c8db0 (d0c8db0), I saw this at boot:

```
[   27.480000] irq 22: nobody cared (try booting with the "irqpoll" option)
[   27.480000] CPU: 0 PID: 839 Comm: obmc-console-se Not tainted 4.10.17-c694e47cbc1efc2bc181b2ac7137dfda04f974c2 #1
[   27.480000] Hardware name: ASpeed SoC
[   27.480000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[   27.480000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
[   27.480000] [<801dbecc>] (dump_stack) from [<8004ca40>] (__report_bad_irq+0x34/0xcc)
[   27.480000] [<8004ca40>] (__report_bad_irq) from [<8004cdcc>] (note_interrupt+0x27c/0x2dc)
[   27.480000] [<8004cdcc>] (note_interrupt) from [<8004a728>] (handle_irq_event_percpu+0x58/0x68)
[   27.480000] [<8004a728>] (handle_irq_event_percpu) from [<8004a770>] (handle_irq_event+0x38/0x4c)
[   27.480000] [<8004a770>] (handle_irq_event) from [<8004d9a8>] (handle_level_irq+0xa4/0x124)
[   27.480000] [<8004d9a8>] (handle_level_irq) from [<80049d94>] (generic_handle_irq+0x30/0x44)
[   27.480000] [<80049d94>] (generic_handle_irq) from [<80049e00>] (__handle_domain_irq+0x58/0xb8)
[   27.480000] [<80049e00>] (__handle_domain_irq) from [<80009488>] (avic_handle_irq+0x74/0x7c)
[   27.480000] [<80009488>] (avic_handle_irq) from [<8000e478>] (__irq_svc+0x58/0x74)
[   27.480000] Exception stack(0x9cbdbc50 to 0x9cbdbc98)
[   27.480000] bc40:                                     80619af8 00000005 00000000 802317e4
[   27.480000] bc60: 80619af8 9e63b8c8 9d0ec600 00000000 00000000 80421380 9e6b9100 9cbdbcc4
[   27.480000] bc80: 00000000 9cbdbca0 802348ac 802317e4 00000013 ffffffff
[   27.480000] [<8000e478>] (__irq_svc) from [<802317e4>] (mem_serial_in+0x0/0x34)
[   27.480000] [<802317e4>] (mem_serial_in) from [<80236294>] (ast_vuart_startup+0x1c/0x4c)
[   27.480000] [<80236294>] (ast_vuart_startup) from [<80234ebc>] (serial8250_startup+0x24/0x30)
[   27.480000] [<80234ebc>] (serial8250_startup) from [<8022e1b4>] (uart_startup.part.0+0x84/0x184)
[   27.480000] [<8022e1b4>] (uart_startup.part.0) from [<8022e31c>] (uart_port_activate+0x68/0x74)
[   27.480000] [<8022e31c>] (uart_port_activate) from [<8022ab8c>] (tty_port_open+0x84/0xd8)
[   27.480000] [<8022ab8c>] (tty_port_open) from [<8022dd7c>] (uart_open+0x40/0x48)
[   27.480000] [<8022dd7c>] (uart_open) from [<80222f68>] (tty_open+0xec/0x5ac)
[   27.480000] [<80222f68>] (tty_open) from [<800e6d14>] (chrdev_open+0x88/0x124)
[   27.480000] [<800e6d14>] (chrdev_open) from [<800df5c8>] (do_dentry_open.constprop.0+0x198/0x2e0)
[   27.480000] [<800df5c8>] (do_dentry_open.constprop.0) from [<800e0830>] (vfs_open+0x50/0x80)
[   27.480000] [<800e0830>] (vfs_open) from [<800ef484>] (path_openat+0x540/0x1090)
[   27.480000] [<800ef484>] (path_openat) from [<800f0b8c>] (do_filp_open+0x74/0xd8)
[   27.480000] [<800f0b8c>] (do_filp_open) from [<800e0d20>] (do_sys_open+0x114/0x1dc)
[   27.480000] [<800e0d20>] (do_sys_open) from [<800e0e10>] (SyS_open+0x28/0x2c)
[   27.480000] [<800e0e10>] (SyS_open) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[   27.480000] handlers:
[   27.480000] [<8021561c>] bt_bmc_irq
[   27.480000] Disabling IRQ #22
```

amboar, 2017-07-09:
Also seen on Witherspoon:

```
[   33.120000] irq 22: nobody cared (try booting with the "irqpoll" option)
[   33.120000] CPU: 0 PID: 935 Comm: obmc-console-se Not tainted 4.10.17-21b60a0716ebca8d2ec600f072a57c78b56ec7a6 #2
[   33.120000] Hardware name: ASpeed SoC
[   33.120000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[   33.120000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
[   33.120000] [<801dbecc>] (dump_stack) from [<8004ca40>] (__report_bad_irq+0x34/0xcc)
[   33.120000] [<8004ca40>] (__report_bad_irq) from [<8004cdcc>] (note_interrupt+0x27c/0x2dc)
[   33.120000] [<8004cdcc>] (note_interrupt) from [<8004a728>] (handle_irq_event_percpu+0x58/0x68)
[   33.120000] [<8004a728>] (handle_irq_event_percpu) from [<8004a770>] (handle_irq_event+0x38/0x4c)
[   33.120000] [<8004a770>] (handle_irq_event) from [<8004d9a8>] (handle_level_irq+0xa4/0x124)
[   33.120000] [<8004d9a8>] (handle_level_irq) from [<80049d94>] (generic_handle_irq+0x30/0x44)
[   33.120000] [<80049d94>] (generic_handle_irq) from [<80049e00>] (__handle_domain_irq+0x58/0xb8)
[   33.120000] [<80049e00>] (__handle_domain_irq) from [<80009488>] (avic_handle_irq+0x74/0x7c)
[   33.120000] [<80009488>] (avic_handle_irq) from [<8000e478>] (__irq_svc+0x58/0x74)
[   33.120000] Exception stack(0x9d891b50 to 0x9d891b98)
[   33.120000] 1b40:                                     00400100 ffff97c1 8060a900 00000000
[   33.120000] 1b60: 00000000 805e3acc 00000008 00000001 9e402600 9d890000 9d890000 9d891bec
[   33.120000] 1b80: 8060a920 9d891ba0 8001c5b4 8000950c 60000113 ffffffff
[   33.120000] [<8000e478>] (__irq_svc) from [<8000950c>] (__do_softirq+0x7c/0x2b4)
[   33.120000] [<8000950c>] (__do_softirq) from [<8001c5b4>] (irq_exit+0x11c/0x13c)
[   33.120000] [<8001c5b4>] (irq_exit) from [<80049e08>] (__handle_domain_irq+0x60/0xb8)
[   33.120000] [<80049e08>] (__handle_domain_irq) from [<80009488>] (avic_handle_irq+0x74/0x7c)
[   33.120000] [<80009488>] (avic_handle_irq) from [<8000e478>] (__irq_svc+0x58/0x74)
[   33.120000] Exception stack(0x9d891c50 to 0x9d891c98)
[   33.120000] 1c40:                                     8061fd84 00000008 00000000 802317e4
[   33.120000] 1c60: 8061fd84 9cfe8990 9cabb400 00000000 00000000 804233a0 9e63dd00 9d891cc4
[   33.120000] 1c80: 00000000 9d891ca0 802321ac 802348a0 00000013 ffffffff
[   33.120000] [<8000e478>] (__irq_svc) from [<802348a0>] (serial8250_do_startup+0xd4/0x6cc)
[   33.120000] [<802348a0>] (serial8250_do_startup) from [<80236294>] (ast_vuart_startup+0x1c/0x4c)
[   33.120000] [<80236294>] (ast_vuart_startup) from [<80234ebc>] (serial8250_startup+0x24/0x30)
[   33.120000] [<80234ebc>] (serial8250_startup) from [<8022e1b4>] (uart_startup.part.0+0x84/0x184)
[   33.120000] [<8022e1b4>] (uart_startup.part.0) from [<8022e31c>] (uart_port_activate+0x68/0x74)
[   33.120000] [<8022e31c>] (uart_port_activate) from [<8022ab8c>] (tty_port_open+0x84/0xd8)
[   33.120000] [<8022ab8c>] (tty_port_open) from [<8022dd7c>] (uart_open+0x40/0x48)
[   33.120000] [<8022dd7c>] (uart_open) from [<80222f68>] (tty_open+0xec/0x5ac)
[   33.120000] [<80222f68>] (tty_open) from [<800e6d14>] (chrdev_open+0x88/0x124)
[   33.120000] [<800e6d14>] (chrdev_open) from [<800df5c8>] (do_dentry_open.constprop.0+0x198/0x2e0)
[   33.120000] [<800df5c8>] (do_dentry_open.constprop.0) from [<800e0830>] (vfs_open+0x50/0x80)
[   33.120000] [<800e0830>] (vfs_open) from [<800ef484>] (path_openat+0x540/0x1090)
[   33.120000] [<800ef484>] (path_openat) from [<800f0b8c>] (do_filp_open+0x74/0xd8)
[   33.120000] [<800f0b8c>] (do_filp_open) from [<800e0d20>] (do_sys_open+0x114/0x1dc)
[   33.120000] [<800e0d20>] (do_sys_open) from [<800e0e10>] (SyS_open+0x28/0x2c)
[   33.120000] [<800e0e10>] (SyS_open) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[   33.120000] handlers:
[   33.120000] [<8021561c>] bt_bmc_irq
[   33.120000] Disabling IRQ #22
```
OpenBMC build:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-88"
VERSION_ID="v1.99.7-88-g5f74f13-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-88"
BUILD_ID="v1.99.7"
```
Kernel:
```
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-21b60a0716ebca8d2ec600f072a57c78b56ec7a6 #2 Fri Jul 7 15:37:28 CDT 2017 armv6l GNU/Linux
```

gtmills, 2018-08-14:
@shenki Old kernel. Still a problem?

closed 2019-04-26 by stale[bot]