openbmc/openbmc #1827: ipmitool sol activate Segmentation fault (core dumped) immediately after reboot of HOST

PavamanSubramaniyam, 2017-06-21:
Connect to a OpenBMC Witherspoon Hardware using IPMITOOL SOL console.
Then execute the reboot command from within the booted Linux Host.
We can see that the IPMI SOL Console immediately seg faults and sol connection is lost.

```
[root@ltc-wspoon4 ~]# uname -a
Linux ltc-wspoon4 4.11.0-8.el7.ppc64le #1 SMP Tue Jun 13 11:43:08 EDT 2017 ppc64le ppc64le ppc64le GNU/Linux
[root@ltc-wspoon4 ~]# cat /etc/os-release
NAME="Red Hat Enterprise Linux Server"
VERSION="7.4 (Pegas)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="7.4"
PRETTY_NAME="Red Hat Enterprise Linux Server 7.4 Beta (Pegas)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:7.4:beta:server"
HOME_URL="https://www.redhat.com/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 7"
REDHAT_BUGZILLA_PRODUCT_VERSION=7.4
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="7.4 Beta"
[root@ltc-wspoon4 ~]# cat /proc/cpuinfo | tail
processor       : 15
cpu             : POWER9 (raw), altivec supported
clock           : 2700.000000MHz
revision        : 1.0 (pvr 004e 0100)

timebase        : 512000000
platform        : PowerNV
model           : 0000000000000000
machine         : PowerNV 0000000000000000
firmware        : OPAL

[root@ltc-wspoon4 ~]# reboot
``` 

```
root@ltc-garri2:~# ipmitool -I lanplus -H <BMC HostName> -P 0penBmc sol activate
[SOL Session operational.  Use ~? for help]
Red Hat Enterprise Linux Server 7.4 Beta (Pegas)
Kernel 4.11.0-8.el7.ppc64le on an ppc64le

ltc-wspoon4 login: [  OK  ] Started Show Plymouth Reboot Screen.
[ 7851.618033] XFS (dm-2): Unmounting Filesystem
[  OK  ] Stopped Availability of block devices.
[  OK  ] Stopped Dynamic System Tuning Daemon.
[  OK  ] Stopped target Network.
         Stopping LSB: Bring up/down networking...
[  OK  ] Started Restore /run/initramfs.
[  OK  ] Stopped LSB: Bring up/down networking.
[  OK  ] Stopped Network Manager Wait Online.
         Stopping Network Manager Wait Online...
         Stopping Network Manager...
[  OK  ] Stopped Network Manager.
[  OK  ] Stopped target Network (Pre).
         Stopping firewalld - dynamic firewall daemon...
[ 7857.984320] Ebtables v2.0 unregistered
[  OK  ] Stopped firewalld - dynamic firewall daemon.
         Stopping D-Bus System Message Bus...
         Stopping Authorization Manager...
[  OK  ] Stopped Authorization Manager.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed RPCbind Server Activation Socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target System Initialization.
[  OK  ] Stopped target Swap.
         Deactivating swap /dev/mapper/rhelp_ltc--wspoon400-swap...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping Apply Kernel Variables...
[  OK  ] Stopped Setup Virtual Console.
         Stopping Setup Virtual Console...
[  OK  ] Stopped target Encrypted Volumes.
         Stopping Update UTMP about System Boot/Shutdown...
         Stopping Load/Save Random Seed...
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Load/Save Random Seed.
         Stopping Security Auditing Service...
[ 7858.838590] audit: type=1305 audit(1498037322.670:353): audit_pid=0 old=1725                                                                                                                                auid=4294967295 ses=4294967295 subj=system_u:system_r:auditd_t:s0 res=1
[  OK  ] Stopped Security Auditing Servic[ 7858.840800] aeudit: type=1131 audit(                                                                                                                               1498037322..670:354): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:s                                                                                                                               /lib/systemd/systemd" hostname=? addr=? terminal=? res=success'

[  OK  ] Stopped Create Volatile Files and[ 7858.842031] a Directories.udit: typ                                                                                                                               audit(1498037322.670:355): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=syste                                                                                                                               m_u:system_r:init_t:s0 msg='unit=systemd-tmpfile
s-setup comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal                                                                                                                               =? res=success'
         Stopping Create Volatile Files and Directories...
[  OK  ] Stopped Import network configura[ 7858.843467] audit: type=1131 audit(1                                                                                                                               498037322.670:356): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:sys                                                                                                                               tem_r:initt_t:s0 msg='unit=rhel-import-state comm="systemd" exe="/usr/lib/ion fr                                                                                                                               om initramsystemd/systemd" hostname=? addr=? terminal=? res=success'
fs.
         Stopping Import network configuration from initramfs...
[  OK  ] Stopped target Local File Systems.
         Unmounting /run/user/0...
         Unmounting Configuration File System...
         Unmounting /boot...
[  OK  ] Stopped Configure read-only root support.
[ 7858.847373] audit: type=1131 audit(1498037322.680:357): pid=1 uid=0 auid=4294                                                                                                                               967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=rhel-readonly c                                                                                                                               omm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=su                                                                                                                               ccess'
         Stopping Configure read-oSegmentation fault (core dumped)
```

PavamanSubramaniyam, 2017-06-21:
```
root@ltc-garri2:~# ipmitool -I lanplus -H <BMC HostName> -P 0penBmc sol activate
[SOL Session operational.  Use ~? for help]
Red Hat Enterprise Linux Server 7.4 Beta (Pegas)
Kernel 4.11.0-8.el7.ppc64le on an ppc64le

ltc-wspoon4 login: [  OK  ] Started Show Plymouth Reboot Screen.
[ 7851.618033] XFS (dm-2): Unmounting Filesystem
[  OK  ] Stopped Availability of block devices.
[  OK  ] Stopped Dynamic System Tuning Daemon.
[  OK  ] Stopped target Network.
         Stopping LSB: Bring up/down networking...
[  OK  ] Started Restore /run/initramfs.
[  OK  ] Stopped LSB: Bring up/down networking.
[  OK  ] Stopped Network Manager Wait Online.
         Stopping Network Manager Wait Online...
         Stopping Network Manager...
[  OK  ] Stopped Network Manager.
[  OK  ] Stopped target Network (Pre).
         Stopping firewalld - dynamic firewall daemon...
[ 7857.984320] Ebtables v2.0 unregistered
[  OK  ] Stopped firewalld - dynamic firewall daemon.
         Stopping D-Bus System Message Bus...
         Stopping Authorization Manager...
[  OK  ] Stopped Authorization Manager.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed RPCbind Server Activation Socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target System Initialization.
[  OK  ] Stopped target Swap.
         Deactivating swap /dev/mapper/rhelp_ltc--wspoon400-swap...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping Apply Kernel Variables...
[  OK  ] Stopped Setup Virtual Console.
         Stopping Setup Virtual Console...
[  OK  ] Stopped target Encrypted Volumes.
         Stopping Update UTMP about System Boot/Shutdown...
         Stopping Load/Save Random Seed...
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Load/Save Random Seed.
         Stopping Security Auditing Service...
[ 7858.838590] audit: type=1305 audit(1498037322.670:353): audit_pid=0 old=1725                                                                                                                                auid=4294967295 ses=4294967295 subj=system_u:system_r:auditd_t:s0 res=1
[  OK  ] Stopped Security Auditing Servic[ 7858.840800] aeudit: type=1131 audit(                                                                                                                               1498037322..670:354): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:s                                                                                                                               /lib/systemd/systemd" hostname=? addr=? terminal=? res=success'

[  OK  ] Stopped Create Volatile Files and[ 7858.842031] a Directories.udit: typ                                                                                                                               audit(1498037322.670:355): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=syste                                                                                                                               m_u:system_r:init_t:s0 msg='unit=systemd-tmpfile
s-setup comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal                                                                                                                               =? res=success'
         Stopping Create Volatile Files and Directories...
[  OK  ] Stopped Import network configura[ 7858.843467] audit: type=1131 audit(1                                                                                                                               498037322.670:356): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:sys                                                                                                                               tem_r:initt_t:s0 msg='unit=rhel-import-state comm="systemd" exe="/usr/lib/ion fr                                                                                                                               om initramsystemd/systemd" hostname=? addr=? terminal=? res=success'
fs.
         Stopping Import network configuration from initramfs...
[  OK  ] Stopped target Local File Systems.
         Unmounting /run/user/0...
         Unmounting Configuration File System...
         Unmounting /boot...
[  OK  ] Stopped Configure read-only root support.
[ 7858.847373] audit: type=1131 audit(1498037322.680:357): pid=1 uid=0 auid=4294                                                                                                                               967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=rhel-readonly c                                                                                                                               omm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=su                                                                                                                               ccess'
         Stopping Configure read-oSegmentation fault (core dumped)
```

PavamanSubramaniyam, 2017-06-21:
The machine is installed with the latest BMC and PNOR Firmware images.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-10"
VERSION_ID="v1.99.7-10-g521a594"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-10"
BUILD_ID="v1.99.7"
root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-witherspoon-v1.17-65-g4e80220
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-87-ge835c2fb7ce1
        hostboot-44d178b
        linux-4.11.2-openpower1-p8f7ea46
        petitboot-v1.4.2-pb361a08
        machine-xml-d0f86cf
        occ-1dc97a6
        hostboot-binaries-24995fc
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%
```

tomjoseph83, 2017-07-24:
@PavamanSubramaniyam What is segfaulting here? I do not see the core dump of net-ipmid or the journal output as proof. 

The core dump message is in the console output.

PavamanSubramaniyam, 2017-08-17:
I am able to re-create this problem again with the witherspoon Hardware connected using out of band ipmitool sol activate.
The question here is why there should be a core dump when we try to connect to a Open BMC box remotely using the out of band ipmitool options.

```
[root@ltc-garri2 ~]# ipmitool -I lanplus -H ltc-wspoon10-bmc.aus.stglabs.ibm.com -P 0penBmc sol activate
[SOL Session operational.  Use ~? for help]

Red Hat Enterprise Linux Server 7.4 Beta (Pegas)
Kernel 4.11.0-22.el7a.ppc64le on an ppc64le

ltc-wspoon10 login: root
Password:
Last login: Thu Aug 17 03:01:27 on hvc0
[root@ltc-wspoon10 ~]# reboot
[  OK  ] Started Show Plymouth Reboot Screen.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Remote File Systems (Pre).
         Stopping Logout off all iSCSI sessions on shutdown...
[  OK  ] Stopped target NFS client services.
         Stopping GSSAPI Proxy Daemon...
[  OK  ] Stopped GSSAPI Proxy Daemon.
[  OK  ] Stopped Logout off all iSCSI sessions on shutdown.
[  OK  ] Stopped Dynamic System Tuning Daemon.
[  OK  ] Stopped target Network.
         Stopping LSB: Bring up/down networking...
[  OK  ] Started Restore /run/initramfs.
[  OK  ] Stopped LSB: Bring up/down networking.
[  OK  ] Stopped Network Manager Wait Online.
         Stopping Network Manager Wait Online...
         Stopping Network Manager...
[  OK  ] Stopped Network Manager.
[  OK  ] Stopped target Network (Pre).
         Stopping firewalld - dynamic firewall daemon...
[  OK  ] Stopped Rollback uncommitted netcf network config change transactions.
         Stopping Rollback uncommitted netcf...rk config change transactions...
[   ***] A stop job is running for firewalld...firewall daemon (13s / 1min 36s)[252803.267381] Ebtables v2.0 unregistered
[  OK  ] Stopped firewalld - dynamic firewall daemon.
         Stopping Authorization Manager...
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped Authorization Manager.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed Virtual machine lock manager socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Closed Open-iSCSI iscsiuio Socket.
[  OK  ] Closed Open-iSCSI iscsid Socket.
[  OK  ] Closed RPCbind Server Activation Socket.
[  OK  ] Closed Virtual machine log manager socket.
[  OK  ] Stopped target System Initialization.
[  OK  ] Stopped target Swap.
         Deactivating swap /dev/mapper/rhelaa_ltc--wspoon10-swap...
[  OK  ] Stopped target Encrypted Volumes.
         Stopping Load/Save Random Seed...
         Stopping Update UTMP about System Boot/Shutdown...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping Apply Kernel Variables...
[  OK  ] Stopped Setup Virtual Console.
         Stopping Setup Virtual Console...
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
         Stopping Security Auditing Service...
[252804.454340] audit: type=1305 audit(1502957354.692:787): audit_pid=0 old=2276 auid=4294967295 ses=4294967295 subj=system_u:system_r:auditd_t:s0 res=1
4.702:788): pid= Security Auditing Service[252804.458791] audit: type=1130. audit(150295735
1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=auditd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[252804.459415] audit: type=1131 audit(1502957354.702:789): pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=auditd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[... 45 lines cut ...]
[252805.073471] systemd-shutdown[1]: Unmounting file systems.
[252805.807287] systemd-shutdown[1]: All filesystems unmounted.
[252805.807466] systemd-shutdown[1]: Deactivating swaps.
[252805.807825] systemd-shutdown[1]: All swaps deactivated.
[252805.807969] systemd-shutdown[1]: Detaching loop devices.
[252805.809447] systemd-shutdown[1]: All loop devices detached.
[252805.809611] systemd-shutdown[1]: Detaching DM devices.
[252805.810962] systemd-shutdown[1]: Detaching DM 253:2.
[252806.075415] shutdown: 7 output lines suppressed due to ratelimiting
[252806.271681] dracut Warning: Killing all remaining processes
dracut Warning: Killing all remaining processes
[252806.882681] XFS (dm-0): Unmounting Filesystem
[252806.976250] dracut Warning: Unmounted /oldroot.
[252807.132029] dracut: Disassembling device-mapper devices
Rebooting.
[252807.327366] kvm: exiting hardware virtualization
[252807.328208] mlx5_core 0033:01:00.1: Shutdown was called
[252807.760639] mlx5_core 0033:01:00.0: Shutdown was called
[252808.189885] sd 1:0:0:0: [sdb] Synchronizing SCSI cache
[252808.190278] mlx5_core 0003:01:00.1: Shutdown was called
[252808.615783] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[252810.591894] mlx5_core 0003:01:00.0: Shutdown was called
[252811.109322] reboot: Restarting system
Segmentation fault (core dumped)
```

PavamanSubramaniyam, 2017-08-17:
The same ipmitool we are using to connect to P9 ZZ Enterprise boxes without any issues.
We are always able to get the SOL access for ZZ P9 Boxes without getting segfault of ipmitool.

I am not able to understand why the same ipmitool out of band command to connect SOL connection works fine on P9 ZZ boxes but segfaults on P9 Open BMC boxes.

tomjoseph83, 2017-08-17:
If you have an ipmitool core dump, please open a ticket against with https://sourceforge.net/p/ipmitool/bugs/

PavamanSubramaniyam, 2017-08-17:
I don't think we should open ticket against ipmitool in the community.
This is a problem specific to Open BMC machines as I had explained since Open BMC machines are not responding/handling the SOL connection properly while using out of band ipmitool commands.

tomjoseph83, 2017-08-17:
Even if a bad response was received by ipmitool it should handle gracefully and should not core-dump.

AlexanderAmelkin, 2018-05-28:
There is an associated bug in ipmitool project:
https://sourceforge.net/p/ipmitool/bugs/495/

AlexanderAmelkin, 2018-05-28:
It is indeed a bug in `ipmitool`, but so far it looks like it is triggered by zero-sized payload returned in an IPMI packet from OpenBMC. See `ipmitool/src/plugins/lanplus/lanplus.c` line 878. The crash in `ipmitool` happens there when the line is executed with `payload_size = 0`. I'm not yet sure if this is a valid or an invalid behavior of OpenBMC.

tomjoseph83, 2018-05-30:
I am not sure if you investigated the same stack as below. In case of SOL, epoll is used and in the case of core-dump there is activity on the fd but there is no IPMI packet received . This is different from the case that OpenBMC has send a valid IPMI packet and payload size is 0. We have a valid use-case to send a SOL payload of size 0. (https://gerrit.openbmc-project.xyz/#/c/6132/)

Thanks @AlexanderAmelkin for updating the issue, let me know if the sourceforge ticket (https://sourceforge.net/p/ipmitool/bugs/495/) needs to be reopened on the github project.

FYI @dkodihal 

```
(gdb) bt
#0  __memmove_avx_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:356
#1  0x0000000000468bea in memmove (__len=18446744073709551612, __src=0x6d17d5 <rsp+21>, __dest=0x6d17c1 <rsp+1>) at /usr/include/x86_64-linux-gnu/bits/string3.h:59
#2  ipmi_lan_poll_single (intf=intf@entry=0x6ceba0 <ipmi_lanplus_intf>) at lanplus.c:884
#3  0x000000000046a718 in ipmi_lan_poll_recv (intf=0x6ceba0 <ipmi_lanplus_intf>) at lanplus.c:919
#4  ipmi_lanplus_recv_sol (intf=0x6ceba0 <ipmi_lanplus_intf>) at lanplus.c:2699
#5  0x0000000000416283 in ipmi_sol_red_pill (intf=intf@entry=0x6ceba0 <ipmi_lanplus_intf>, instance=instance@entry=1) at ipmi_sol.c:1652
#6  0x000000000041674c in ipmi_sol_activate (intf=intf@entry=0x6ceba0 <ipmi_lanplus_intf>, looptest=looptest@entry=0, interval=interval@entry=0, instance=1)
    at ipmi_sol.c:1862
#7  0x0000000000417091 in ipmi_sol_main (intf=0x6ceba0 <ipmi_lanplus_intf>, argc=<optimized out>, argv=<optimized out>) at ipmi_sol.c:2014
#8  0x000000000044015b in ipmi_main (argc=11, argv=<optimized out>, cmdlist=cmdlist@entry=0x6c2560 <ipmitool_cmd_list>, intflist=intflist@entry=0x0) at ipmi_main.c:1003
#9  0x0000000000403ba0 in main (argc=<optimized out>, argv=<optimized out>) at ipmitool.c:135
```

```
               if (rsp && extra_data_length) {
                        rsp->data_len = extra_data_length;
                        memmove(rsp->data, rsp->data + offset, extra_data_length);
```

AlexanderAmelkin, 2018-05-30:
@tomjoseph83, thanks for our comments. I will investigate further. No need to reopen the sf issue on github. I'm working on importing the tickets in bulk mode.

closed 2019-04-27 by stale[bot]