openbmc/openbmc #1828: [Barreleye] question: where critical_lower and critical_upper located or would be located?

yadron, 2017-06-21:
If I run old format request for sensor value, then critical_lower 0 and critical_upper 0 exists:

```
curl -b cjar -k https://172.17.27.28/org/openbmc/sensors/temperature/ambient
{
  "data": {
    "critical_lower": 0, 
    "critical_upper": 0, 
    "emergency_enabled": 0, 
    "error": 0, 
    "filename": "/sys/class/hwmon/hwmon1/temp1_input", 
    "offset": 0, 
    "scale": -3, 
    "threshold_state": "NORMAL", 
    "thresholds_enabled": 0, 
    "units": "C", 
    "value": 31500, 
    "warning_lower": 0, 
    "warning_upper": 0, 
    "worst_threshold_state": "NORMAL"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

If I run new xyz style request, it doesn't:

```
curl -b cjar -k https://172.17.27.28/xyz/openbmc_project/sensors/temperature/ambient
{
  "data": {
    "Scale": -3, 
    "Unit": "xyz.openbmc_project.Sensor.Value.Unit.DegreesC", 
    "Value": 31500
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

Question. Does it mean that these fields will be added to xyz and DBUS later? Could you please answer where it will be added? What is the plan for naming (both what is the name going to be in DBUS and what is naming for RestAPI/curl)?

yadron, 2017-06-21:
Also I've noted that from the system point of view min/max already exists:

```
root@barreleye:~# cat /sys/class/hwmon/hwmon1/device/in2_min 
0
root@barreleye:~# cat /sys/class/hwmon/hwmon1/device/in2_max 
2046
```

williamspatrick, 2017-06-28:
@bradbishop - This is all handled through a YAML file to phosphor-hwmon, right?

bradbishop, 2017-06-28:
correct.  Here is an example:  https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-palmetto/recipes-phosphor/sensors/phosphor-hwmon%25/obmc/hwmon/ahb/apb/i2c%401e78a000/i2c-bus%4040/rtc%4068.conf

bradbishop, 2017-06-28:
I guess technically its not a yaml file....its a runtime config file ( the thresholds are pulled from the environment).

AlexanderAmelkin, 2018-03-05:
What happens if some sensors goes over its WARNHI threshold?
IPMI specifies 6 thresholds (upper/lower non-critical/critical/non-recoverable), each of which can be 'going low' or 'going high' in terms of SEL. Then a PEF entry can be created to react to those threshold crossings. What is the paradigm in OpenBMC for reacting to critical sensor events and how it correlates to IPMI? What (sub)system is responsible (or planned to be responsible) for shutting down a critically overheating system?

closed 2019-04-26 by stale[bot]