openbmc/openbmc #1829: Is the /xyz/openbmc_project/inventory/system endpoint removed?

whowutwut, 2017-06-21:
### ITC LEVEL 

Doing a GET on the `/xyz/openbmc_project/inventory/system` endpoint, returns **Object not found** error.  
```
#  curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://172.20.253.103//xyz/openbmc_project/inventory/system
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system"
  },
  "message": "404 Not Found",
  "status": "error"
}
```

ITC level machine preloaded with the following FW versions:
```
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.5"
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.16_1.24
```

### EUH Level 

Our EUH machine with FW level:
```
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-130"
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.17_1.20
```
Shows the following: 
```
$ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://10.6.9.254/xyz/openbmc_project/inventory/system
{
  "data": {
    "BuildDate": "",
    "Cached": 0,
    "FieldReplaceable": 0,
    "Manufacturer": "",
    "Model": "0000000000000000",
    "PartNumber": "",
    "Present": 1,
    "PrettyName": "",
    "SerialNumber": "0000000000000000"
  },
  "message": "200 OK",
  "status": "ok"
}
```

### w51 

But what concerns me is the w51 machine that I just ran a curl command again, also returning 404.. and it's a later version of the FW.

```
# ./show_firmware_level 9.3.181.64
root@9.3.181.64's password:
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-3"
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.17_1.33
```

```
# curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://9.3.181.64/xyz/openbmc_project/inventory/system
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/inventory/system"
  },
  "message": "404 Not Found",
  "status": "error"
}
```

Trying to understand if this was removed and why....?  
@bradbishop @williamspatrick @geissonator  Not sure who to direct this to...?

gkeishin, 2017-06-22:
I can still see this one..
```
Executing: GET  /xyz/openbmc_project/inventory/system

/xyz/openbmc_project/inventory/system
{
    "BuildDate": "",
    "Cached": 0,
    "FieldReplaceable": 0,
    "Manufacturer": "",
    "Model": "0000000000000000",
    "PartNumber": "",
    "Present": 1,
    "PrettyName": "",
    "SerialNumber": "0000000000000000"
}

```

Check if the Host is running
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
You'll only see that when the Host boots up.

whowutwut, 2017-06-22:
Oh ok. That might be the reason. Will check. 

So this may be a problem for us since before the host is booted we are trying to obtain some unique identifier from the BMC. We planned to take the model and serial combination.  In this discovery process it's not required to have the host state booted yet.. 

Is there some info that we can obtain when the host is powered off and only the BMC had powered that would be unique?

gkeishin, 2017-06-22:
@whowutwut At the moment at poweroff if there were not previous power on, you could see something like this
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

/xyz/openbmc_project/inventory/enumerate
{
    "/xyz/openbmc_project/inventory/system/chassis": {
        "AirCooled": 1,
        "WaterCooled": 0
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc": {
        "BuildDate": "",
        "FieldReplaceable": 0,
        "Manufacturer": "IBM",
        "Model": "",
        "PartNumber": "01DH051",
        "Present": 1,
        "PrettyName": "BMC PLANAR  ",
        "SerialNumber": "000000000000"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc/ethernet": {
        "FieldReplaceable": 0,
        "MACAddress": "00:00:00:00:00:00",
        "Present": 1,
        "PrettyName": ""
    }
}
```

Shall we close this one now ?

mdmillerii, 2017-06-22:
@whowutwut there is an issue to expose a uuid from vpd via /etc/machine-id and on to dbus.
@anoo1 can you link the issue?

whowutwut, 2017-06-22:
@gkeishin  Yea, I see the motherboard/boxelder... but this seems to be information about the motherboard and seems like none of it is filled in.  Is that something at MFG need to burn in? 

@mdmillerii  It might be #479 , do you know when this would be exposed in a REST call?

gkeishin, 2017-06-22:
@whowutwut The boxelder is currently not programmed in VPD so those are empty for now.

whowutwut, 2017-07-06:
@gkeishin Do you know what it should be programmed with later on?   I'm trying to find some unique way to identify the nodes when we are looking at the BMCs, in the past we used the combination of `node-<model>-<serial>` so this string is unique.  Here even if we used the boxelder PartNumber/Serial, the serial is currently zeros and so until that gets filled in, we can't rely on this as being unique.  No response about the uuid exposed in REST to know if this is a viable option.  

@zet809 FYI

gkeishin, 2017-07-06:
@dkodihal  any info this question above

williamspatrick, 2017-07-06:
@whowutwut - We are unable to get the system serial number until the host has been booted once.  So, after a factory reset we are certain to not be have that.  I would expect most machines to ship from manufacturing having been booted once and have inventory persisted, but I'm not certain on their process yet as it hasn't been defined.

Relying on the boxelder PartNumber/Serial or UUID would be fine, I think, but those are not constant in the event of a field replacement of the boxelder card.  Is this ok?

whowutwut, 2017-07-20:
Thanks, seems like without the first IPL boot, we should only rely on the boxelder information.  I confirmed with MFG that the STC machines should show up with the MAC/serial burned in... (hopefully)....  so xCAT will modify the endpoints and where to look for this information.

closed 2017-07-20 by whowutwut