openbmc/openbmc #1831: Can not get network information & How to set IP Address/Netmask/Gateway/VLAN?

xuweibj, 2017-06-22:
1. Currently, can not get network information:
```
# curl -c cjar -b cjar -k -H "Content-Type: application/json" https://x.x.x.x/xyz/openbmc_project/network/enumerate
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/network"
  },
  "message": "404 Not Found",
  "status": "error"
}
```

2. How to set IP Address/Netmask/Gateway/VLAN?

I have got information from https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Network , but could you please help to provide exact interface of setting?

gkeishin, 2017-06-22:
I dont think the network code to enable xyz domain interface has gone in..   @ratagupt Can clarify more on this one..

mine260309, 2017-06-27:
Current OpenBMC is still using `org.openbmc.NetworkManager` to manage network, see https://github.com/openbmc/docs/blob/master/dbus-interfaces.md#orgopenbmcnetworkmanager for reference.
It provides functions to get/set addresses.

E.g. to get the network address, invoke `GetAddress4` method:
```
curl -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["eth0"]}' https://bmc/org/openbmc/NetworkManager/Interface/action/GetAddress4
```

xuweibj, 2017-06-28:
Hi @mine260309 , thank you for your comment.

I have tried the method you provide, it worked well.
But I still have questions as below:
1. How could I know what NIC name I should use in request? In your comment use ``eth0``, but I think maybe need use ``eth1`` or other to get other machines. Is there interface to get NIC name or list?
2. The result I got for ``curl -b cjar -k -H "Content-Type: application/json" -X POST -d '{"data": ["eth0"]}' https://bmc/org/openbmc/NetworkManager/Interface/action/GetAddress4`` is below:
```
{
  "data": [
    2,
    8,
    "10.6.7.254",
    "10.0.0.101"
  ],
  "message": "200 OK",
  "status": "ok"
}
```
I think that ``8 -- Netmak``, ``10.6.7.254 -- BMC IP``, ``10.0.0.101 -- Gateway``. Is that right? And what 2 represent?
3. If I want to set IP, how to define the ``data``?

bradbishop, 2017-06-29:
@xuweibj This is the old v1 API, which will be changing in the very near future.  Please don't code to this.

bradbishop, 2017-06-29:
@ratagupt please provide some instructions for @xuweibj so she can write her code, preferably in the form of a documentation update to https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Network/README.md

xuweibj, 2017-07-03:
Hi, is there any progress about this issue?

whowutwut, 2017-07-05:
@ratagupt Any updates?  

I've opened another issue #1899 to track the URGENT issue of providing in-band VLAN tagging support.  This issue can continue tracking the out of band Network Configuration via REST.

ratagupt, 2017-07-06:
@whowutwut : we don't have vlan support till now,Will make it on priority.for other networking code as I had not enabled it through recipe, so xyz namespace will not work.

whowutwut, 2017-07-06:
@ratagupt Great, thanks.  The #1899 is actually the hot item now, to be able to set VLAN inband, so having out of band is OK but it's not the immediate request. The other IP/NETMASK/GATEWAY can all be set inband which is what we use during initial discovery and configuration of the BMC network properties.  Having ability to set out of band rest is  a secondary priority.  This is why I opened the 1899 issue.

xuweibj, 2017-07-20:
Hi, I noticed that #1678 had been closed. So is there exact interface to set IP/NETMASK/GATEWAY now? May I get it now?

whowutwut, 2017-07-25:
@ratagupt  From the documentation on the [Network](https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Network/README.md) I'm not able to understand how we would consume the VLAN properties just to display it....   Under the VLAN.interface.yaml file, it shows "InterfaceName" and "Id". 

 Is there going to be multiple vlan attributes, or a single one for the main network device?  Here's what I mean.... 

Single....
```
"/xyz/openbmc_project/network/vlan": {
    InterfaceName: eth0,
    Id: 123
}
```

Or Multiple....
```
"/xyz/openbmc_project/network/vlan/<some id>": {
    InterfaceName: eth0,
    Id: 123
}
"/xyz/openbmc_project/network/vlan/<some id>": {
    InterfaceName: eth0,
    Id: 456
}
```

* Since this is not documented, it's impossible for us to code against the API without implementation already there..   
* without an API clearly defined that means the  API can change at will, breaking any existing code already developed.

ratagupt, 2017-07-27:
Interface which you are looking for is to configure the vlan?

xuweibj, 2017-07-28:
@ratagupt We also need the interface to configure IP, netmask and gateway.

gurevichmark, 2017-07-31:
Using the latest BMC build `"Version": "ibm-v1.99.8-98-g06734f31-r2-0-g752ced3`

The way to set Gateway property, for example, is:
```
 curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data": "1.2.3.4"}' https://$BMCIP/xyz/openbmc_project/network/eth0/ipv4/26238a40/attr/Gateway
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}[root@stratton01 ob910.1730.20170728n]#
```

Enumerate now shows:
```
"/xyz/openbmc_project/network/eth0/ipv4/26238a40": {
      "Address": "10.6.5.254",
      "Gateway": "1.2.3.4",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 8,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
```

gurevichmark, 2017-07-31:
Should this issue be closed, since the implementation work is being done under issue #3339

whowutwut, 2017-08-01:
Yes, I think it can be closed, will let @xuweibj  close it out.  The OpenBMC OS is planned to be upgraded to 1.99.8 in 1732A driver, which ETA is this Friday.

xuweibj, 2017-08-01:
OK. Close this issue.

closed 2017-08-01 by xuweibj

ratagupt, 2017-08-01:
@xuweibj :- The gateway which is in ipv4 object is having the network gateway for the ip addess,If you want to set the default gateway then you need to look at the config object.
/xyz/openbmc_project/network/config.
It is having two properties 1) hostname 2) defaultgateway.