openbmc/openbmc #1832: Host OS hung after injecting L3 FIR checkstop through cronus putscom operations [witherspoon]]

SrideviRamesh, 2017-06-22:
Issue:

Host OS hung after injecting L3 FIR checkstop through cronus putscom operations

System Info:

```
# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-25"
VERSION_ID="v1.99.7-25-g00714a4"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-25"
BUILD_ID="v1.99.7"
root@witherspoon:~# 

```

Test Procedure:

1. HOST OS should be running before the error injection.
2. Inject below error through BMC cronus setup

```
putscom pu.ex 10011800 8000000000000000 -p0 -c9

```

Attached ffdc, dmesg, journal, esel logs
[20170622060747335641_MyFFDCLogs.zip](https://github.com/openbmc/openbmc/files/1094568/20170622060747335641_MyFFDCLogs.zip)
[esel2.out.txt](https://github.com/openbmc/openbmc/files/1094571/esel2.out.txt)
[witherspoon_dmesg.txt](https://github.com/openbmc/openbmc/files/1094572/witherspoon_dmesg.txt)
[witherspoon_journal.txt](https://github.com/openbmc/openbmc/files/1094573/witherspoon_journal.txt)

gkeishin, 2017-06-22:
So the interrupt is recieved and it rebooting the Host

```
Apr 21 09:21:11 witherspoon phosphor-gpio-monitor[1594]: GPIO line altered
Apr 21 09:21:12 witherspoon systemd[1]: Created slice system-openpower\x2ddebug\x2dcollector\x2dcheckstop.slice.
Apr 21 09:21:12 witherspoon systemd[1]: Starting OpenPOWER debug data collector for host checkstop...
Apr 21 09:21:12 witherspoon checkstop_app[2279]: Host checkstop condition detected
Apr 21 09:21:12 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Apr 21 09:21:12 witherspoon systemd[1]: Reached target Quiesce Target.
Apr 21 09:21:12 witherspoon systemd[1]: Reached target Host instance 0 crashed.
Apr 21 09:21:12 witherspoon phosphor-host-state-manager[1038]: Auto reboot enabled. Reducing HOST BOOTCOUNT by 1.
Apr 21 09:21:12 witherspoon sensor_manager2.py[940]: Setting boot count to 2
Apr 21 09:21:12 witherspoon phosphor-host-state-manager[1038]: Beginning reboot...
Apr 21 09:21:12 witherspoon phosphor-host-state-manager[1038]: Host State transaction request
```
This is an indication the Host still in booting process.. 

```
Apr 21 09:26:13 witherspoon ipmid[1068]: A host system event was received
Apr 21 09:27:18 witherspoon timemanager[1073]: Request to SET Host time
```
Are we running this manually or via automation @SrideviRamesh 

```
Apr 21 09:31:03 witherspoon dropbear[2790]: Exit (root): Exited normally
```

gkeishin, 2017-06-22:
By the way, this driver has a kernel bug also https://github.com/openbmc/openbmc/issues/1830 which is now fixed in latest build. Probably try there again

sivassrr, 2017-06-22:
@SrideviRamesh Have we tried the same using Host OS checkstop error injection way? Is it working.

lkammath, 2017-06-23:
George,

I am not sure what kernel bug has to do with the error isolation that PRD does. I would recommend some one from PRD comment on this

thanks,
Lakshmi

lkammath, 2017-06-23:
@gkeishin Even injection through Host is not having a PRD isolation

http://ausgsa.ibm.com//gsa/ausgsa-p10/03/indiateam04/gkeishin/openbmc-test-automation/logs/20170623071722796027_Myffdc/20170623071722796027_MyFFDCLogs/20170623071722796027_esel.text

spinler, 2017-06-23:
Wrong repository for tracking PRD problems.  This is for BMC problems only.

closed 2017-06-23 by spinler

kachini, 2017-11-13:
folks, what tool have been used to generate esel.txt files?

gkeishin, 2017-11-13:
eSEL.pl file and op binaries for the text errorlog conversion from raw eSEL data @kachini

gkeishin, 2017-11-13:
You can enumerate and grab the esel via REST

kachini, 2017-11-13:
@gkeishin do you mean just read with something like this: "curl -b cjar -k https://bmc/xyz/openbmc_project/logging/entry/1    ?
It generates very limited output but i need info to debug a checkstop . How to get the "op binaries" ?