openbmc/openbmc #1833: ASpeed SoC : w1_slave_driver Device registration Failed  [Ziaus]

gkeishin, 2017-06-22:
This failure continuously flooding the dmesg log

Driver info:
```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-28"
VERSION_ID="v1.99.7-28-g65333c0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-28"
BUILD_ID="v1.99.7"
root@zaius:~# uname -a
Linux zaius 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1 Thu Jun 22 06:21:55 UTC 2017 armv6l GNU/Linux
root@zaius:~#
```
````
[  867.820000] ------------[ cut here ]------------
[  867.820000] WARNING: CPU: 0 PID: 413 at /tmp/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x68/0x88
[  867.850000] sysfs: cannot create duplicate filename '/bus/w1/devices/00-f00000000000'
[  867.850000] CPU: 0 PID: 413 Comm: w1_bus_master4 Tainted: G        W       4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1
[  867.850000] Hardware name: ASpeed SoC
[  867.850000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[  867.850000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
[  867.850000] [<801dbecc>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
[  867.890000] [<80018d54>] (__warn) from [<80018dc0>] (warn_slowpath_fmt+0x44/0x4c)
[  867.890000] [<80018dc0>] (warn_slowpath_fmt) from [<80145d54>] (sysfs_warn_dup+0x68/0x88)
[  867.890000] [<80145d54>] (sysfs_warn_dup) from [<80146040>] (sysfs_do_create_link_sd+0x94/0xa4)
[  867.890000] [<80146040>] (sysfs_do_create_link_sd) from [<801460a4>] (sysfs_create_link+0x34/0x44)
[  867.890000] [<801460a4>] (sysfs_create_link) from [<80240828>] (bus_add_device+0x12c/0x1b8)
[  867.890000] [<80240828>] (bus_add_device) from [<8023eacc>] (device_add+0x2a0/0x59c)
[  867.940000] [<8023eacc>] (device_add) from [<8023edec>] (device_register+0x24/0x28)
[  867.940000] [<8023edec>] (device_register) from [<802da924>] (w1_attach_slave_device+0x1b0/0x3d4)
[  867.940000] [<802da924>] (w1_attach_slave_device) from [<802dac9c>] (w1_slave_found+0xb8/0xbc)
[  867.940000] [<802dac9c>] (w1_slave_found) from [<802db560>] (w1_search+0x258/0x2b8)
[  867.940000] [<802db560>] (w1_search) from [<802dc83c>] (w1_search_devices+0x50/0x54)
[  867.940000] [<802dc83c>] (w1_search_devices) from [<802db640>] (w1_search_process_cb+0x80/0x12c)
[  867.940000] [<802db640>] (w1_search_process_cb) from [<802db868>] (w1_process+0xf4/0x13c)
[  868.000000] [<802db868>] (w1_process) from [<800335ec>] (kthread+0x124/0x164)
[  868.000000] [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
[  868.000000] ---[ end trace 40e2923efc6aa96d ]---
[  868.000000] w1_slave_driver 00-f00000000000: Device registration [00-f00000000000] failed. err=-17
[  868.000000] w1_master_driver w1_bus_master4: w1_attach_slave_device: Attaching 00-f00000000000 failed.
```

- [x] dmesg 
[dmesg_v1.99.7-28-g65333c0_w1_slave_driver_ZIAUS.txt](https://github.com/openbmc/openbmc/files/1094909/dmesg_v1.99.7-28-g65333c0_w1_slave_driver_ZIAUS.txt)

gkeishin, 2017-06-22:
LOG: 
[dmesg_v1.99.7-28-g65333c0_w1_slave_driver_ZIAUS.txt](https://github.com/openbmc/openbmc/files/1094920/dmesg_v1.99.7-28-g65333c0_w1_slave_driver_ZIAUS.txt)

GHF, 2017-06-26:
This happens when the 1-wire buses (schematic nets `FM_SENSOR_TEMP{0,1,2,3}`) are not pulled high and the kernel driver identifies that as a device presence. You can probe your motherboard if the hardware is in front of you. I believe this is resolved on EVT2 and higher hardware.

gkeishin, 2017-09-18:
Closing this.. will open if seen ahead

closed 2017-09-18 by gkeishin