openbmc/openbmc #1835: Fan Control - Add 'delay' parameter on events

msbarth, 2017-06-22:
Until OCC goes "Active", a group of sensors provided by OCC will not exist and should be delayed from having the fan control application matching on those dbus signals.
Two event parameters for a 'delay' will be needed:
- Property changed delay (listen for a property change on startup that triggers fan control to start matching
- Time delay (allow a time based delay for instances where property change signal is not available)

bjwyman, 2017-07-12:
STATUS:
Actually had a start Tuesday. Discussion with @msbarth. Started updating example/events.yaml.
07/12 - Added occ active condtion check to events.yaml.
- Started updating functor.hpp for handling a condition, and created conditions.hpp.
- Simple updates to YAML parser, but apparently not simple enough, bitbake build fails.

geissonator, 2017-07-13:
Need task list

bjwyman, 2017-07-13:
- [ ] Update events YAML file(s) to have conditions section.
 * Add condition to check for OCC status active.
- [ ] Update fan control code to create and use condition checking method.
- [ ] Update fan control YAML parser to handle conditions for fan_zone_defs.cpp

msbarth, 2017-07-26:
- [ ] Review functional implementation of provided code
- [ ] Add 'condition' section to events yaml supporting 1-to-many property changed conditions
- [ ] Determine yaml and generated source layout of using conditions within an event
- [ ] Read the conditions' properties and subscribe to their signal changes prior to init'ing any event
- [ ] Only init events when no condition is provided or 'condition' is in TRUE state
- [ ] When condition is TRUE, get the event group's properties, subscribe to their signal changes, and run the event action
- [ ] When condition is FALSE, unsubscribe from event group's signal changes and remove from property cache

# **STATUS**
7/26
Caught up on things while out and had meeting(1.5hr) with Todd over OpenBMC.
Walked thru #1835 handoff with Brandon, begin reviewing #1835 changes for reuse after acquiring.

No scrum, demo meeting, retrospective discussion, and 0x3030 support discussion(3.5hr total)
Determine final design for #1835 based on what's intended and code provided.
Walk thru provided #1835 code to determine re-usability with decided design.
Begin implementation layout framework(yaml, generated code structure, etc...)

Blockers -

rfrandse, 2017-08-08:
<https://gerrit.openbmc-project.xyz/5859> Update parser to support optional preconditions
Resolves  #1835 Fan Control - Add 'delay' parameter on events

closed 2017-08-14 by williamspatrick