openbmc/openbmc #1836: [Palmetto] fsi-master-gpio: probe of gpio-fsi failed with error -16

swe12345, 2017-06-23:
fsi-master-gpio: probe of gpio-fsi failed with error -16

Driver Details
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-29"
VERSION_ID="v1.99.7-29-ga20607b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-29"
BUILD_ID="v1.99.7"
```
```
root@palmetto:~# cat /etc/timestamp
20170622092455
```
```
Apr 20 17:29:31 palmetto kernel[791]: [    5.000000] Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
Apr 20 17:29:31 palmetto kernel[791]: [    5.000000] console [ttyS4] disabled
Apr 20 17:29:31 palmetto kernel[791]: [    5.000000] 1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 23, base_baud = 1500000) is a 16550A
Apr 20 17:29:31 palmetto kernel[791]: [    5.470000] console [ttyS4] enabled
Apr 20 17:29:31 palmetto kernel[791]: [    5.470000] 1e787000.vuart: ttyS0 at MMIO 0x1e787000 (irq = 22, base_baud = 1500000) is a 16550A
Apr 20 17:29:31 palmetto kernel[791]: [    5.470000] timeriomem_rng 1e6e2000.syscon:hwrng@1e6e2078: timeriomem_rng: resource is missing
Apr 20 17:29:31 palmetto kernel[791]: [    5.470000] brd: module loaded
Apr 20 17:29:31 palmetto kernel[791]: [    5.610000] loop: module loaded
Apr 20 17:29:31 palmetto kernel[791]: [    5.610000] aspeed-lpc-ctrl 1e789080.lpc-ctrl: Loaded at 0x98000000 (0x01000000)
Apr 20 17:29:31 palmetto kernel[791]: [    5.640000] aspeed-smc 1e620000.fmc: Using IRQ 18
Apr 20 17:29:31 palmetto kernel[791]: [    5.640000] aspeed-smc 1e620000.fmc: DMA support disabled.
Apr 20 17:29:31 palmetto kernel[791]: [    5.650000] aspeed-smc 1e620000.fmc: mx25l25635e (32768 Kbytes)
Apr 20 17:29:31 palmetto kernel[791]: [    5.650000] aspeed-smc 1e620000.fmc: CE0 window [ 0x20000000 - 0x22000000 ] 32MB
Apr 20 17:29:31 palmetto kernel[791]: [    5.650000] aspeed-smc 1e620000.fmc: CE1 window [ 0x22000000 - 0x26000000 ] 64MB
Apr 20 17:29:31 palmetto kernel[791]: [    5.670000] 5 ofpart partitions found on MTD device bmc
Apr 20 17:29:31 palmetto kernel[791]: [    5.680000] Creating 5 MTD partitions on "bmc":
Apr 20 17:29:31 palmetto kernel[791]: [    5.690000] 0x000000000000-0x000000060000 : "u-boot"
Apr 20 17:29:31 palmetto kernel[791]: [    5.690000] 0x000000060000-0x000000080000 : "u-boot-env"
Apr 20 17:29:31 palmetto kernel[791]: [    5.710000] 0x000000080000-0x0000004c0000 : "kernel"
Apr 20 17:29:31 palmetto kernel[791]: [    5.710000] 0x0000004c0000-0x000001c00000 : "rofs"
Apr 20 17:29:31 palmetto kernel[791]: [    5.730000] 0x000001c00000-0x000002000000 : "rwfs"
Apr 20 17:29:31 palmetto kernel[791]: [    5.730000] aspeed-smc 1e630000.spi: Error applying setting, reverse things back
Apr 20 17:29:31 palmetto kernel[791]: [    5.770000] aspeed-smc 1e630000.spi: mx25l25635e (32768 Kbytes)
Apr 20 17:29:31 palmetto kernel[791]: [    6.560000] libphy: Fixed MDIO Bus: probed
Apr 20 17:29:31 palmetto kernel[791]: [    6.560000] ftgmac100 1e660000.ethernet: Read MAC address a0:42:3f:25:16:fe from chip
Apr 20 17:29:31 palmetto kernel[791]: [    6.560000] ftgmac100 1e660000.ethernet: Using NCSI interface
Apr 20 17:29:31 palmetto kernel[791]: [    6.560000] ftgmac100 1e660000.ethernet eth0: irq 20, mapped at f0060000
Apr 20 17:29:31 palmetto kernel[791]: [    6.560000] fsi-master-gpio gpio-fsi: failed to get trans gpio
Apr 20 17:29:31 palmetto kernel[791]: [    6.670000] fsi-master-gpio: probe of gpio-fsi failed with error -16
Apr 20 17:29:31 palmetto kernel[791]: [    6.670000] i2c /dev entries driver
Apr 20 17:29:31 palmetto kernel[791]: [    6.670000] at24 0-0050: 32768 byte 24c256 EEPROM, writable, 64 bytes/write
Apr 20 17:29:31 palmetto kernel[791]: [    6.710000] rtc-ds1307 0-0068: rtc core: registered ds3231 as rtc0
```

[journalctl_23jun.txt](https://github.com/openbmc/openbmc/files/1096820/journalctl_23jun.txt)

amboar, 2017-06-26:
Sounds like an incomplete gpio description in the devicetree.

amboar, 2017-06-26:
Okay, looking at `aspeed-bmc-opp-palmetto.dts` it seems that all the properties are present. It's also possible another driver has already claimed the pin. Can you please run `cat /sys/kernel/debug/pinctrl/1e6e2000.syscon:pinctrl@1e6e2000/pinmux-pins       | grep -v UNCLAIMED` and paste the output here?

swe12345, 2017-06-28:
Observing on Barreleye system as well
Driver Details:
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-61"
VERSION_ID="v1.99.7-61-gb42df4d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-61"
BUILD_ID="v1.99.7"
```
journalctl contains
```
Apr 28 13:32:25 barreleye systemd[1]: phosphor-gpio-monitor@checkstop.service: Failed with result 'core-dump'.
```

amboar, 2017-06-29:
@swe12345 can you please provide the output I requested in https://github.com/openbmc/openbmc/issues/1836#issuecomment-310943209

swe12345, 2017-06-30:
Sorry missed the comment. 
```
root@palmetto:~# cat /sys/kernel/debug/pinctrl/1e6e2000.syscon:pinctrl@1e6e2000/pinmux-pins | grep -v UNCLAIMED
Pinmux settings per pin
Format: pin (name): mux_owner|gpio_owner (strict) hog?
pin 0 (D6): GPIO 1e780000.gpio:292
pin 1 (B5): GPIO 1e780000.gpio:293
pin 2 (A4): GPIO 1e780000.gpio:294
pin 4 (C5): GPIO 1e780000.gpio:296
pin 5 (B4): GPIO 1e780000.gpio:297
pin 6 (A3): GPIO 1e780000.gpio:298
pin 9 (J20): GPIO 1e780000.gpio:301
pin 10 (H18): GPIO 1e780000.gpio:302
pin 14 (H20): GPIO 1e780000.gpio:306
pin 15 (E18): GPIO 1e780000.gpio:307
pin 20 (D4): GPIO 1e780000.gpio:312
pin 21 (C3): GPIO 1e780000.gpio:313
pin 22 (B2): GPIO 1e780000.gpio:314
pin 23 (A1): GPIO 1e780000.gpio:315
pin 24 (A18): GPIO 1e780000.gpio:316
pin 25 (D16): GPIO 1e780000.gpio:317
pin 32 (D15): GPIO 1e780000.gpio:324
pin 33 (C15): GPIO 1e780000.gpio:325
pin 41 (B19): GPIO 1e780000.gpio:333
pin 44 (B18): GPIO 1e780000.gpio:336
pin 45 (A19): GPIO 1e780000.gpio:337
pin 47 (C17): GPIO 1e780000.gpio:339
pin 51 (C13): GPIO 1e780000.gpio:343
pin 52 (B13): GPIO 1e780000.gpio:344
pin 53 (Y21): GPIO 1e780000.gpio:345
pin 54 (AA22): device 1e6e2000.syscon:pinctrl@1e6e2000 function FLBUSY group FLBUSY
pin 55 (U18): device 1e6e2000.syscon:pinctrl@1e6e2000 function FLWP group FLWP
pin 56 (A8): GPIO 1e780000.gpio:348
pin 57 (C7): GPIO 1e780000.gpio:349
pin 58 (B7): GPIO 1e780000.gpio:350
pin 62 (A6): GPIO 1e780000.gpio:354
pin 63 (E7): GPIO 1e780000.gpio:355
pin 75 (J3): GPIO 1e780000.gpio:367
pin 76 (T4): device 1e6e2000.syscon:pinctrl@1e6e2000 function VGAHS group VGAHS
pin 77 (U2): device 1e6e2000.syscon:pinctrl@1e6e2000 function VGAVS group VGAVS
pin 78 (T2): device 1e6e2000.syscon:pinctrl@1e6e2000 function DDCCLK group DDCCLK
pin 79 (T1): device 1e6e2000.syscon:pinctrl@1e6e2000 function DDCDAT group DDCDAT
pin 80 (E3): device 1e78a140.i2c-bus function I2C5 group I2C5
pin 81 (D2): device 1e78a140.i2c-bus function I2C5 group I2C5
pin 82 (C1): device 1e78a180.i2c-bus function I2C6 group I2C6
pin 83 (F4): device 1e78a180.i2c-bus function I2C6 group I2C6
pin 84 (E2): device 1e78a1c0.i2c-bus function I2C7 group I2C7
pin 85 (D1): device 1e78a1c0.i2c-bus function I2C7 group I2C7
pin 86 (G5): device 1e78a300.i2c-bus function I2C8 group I2C8
pin 87 (F3): device 1e78a300.i2c-bus function I2C8 group I2C8
pin 128 (D3): device 1e78a0c0.i2c-bus function I2C3 group I2C3
pin 129 (C2): device 1e78a0c0.i2c-bus function I2C3 group I2C3
pin 130 (B1): device 1e78a100.i2c-bus function I2C4 group I2C4
pin 131 (F5): device 1e78a100.i2c-bus function I2C4 group I2C4
pin 135 (H1): GPIO 1e780000.gpio:427
pin 140 (V21): GPIO 1e780000.gpio:432
pin 141 (W22): GPIO 1e780000.gpio:433
pin 152 (A12): device 1e660000.ethernet function RMII1 group RMII1
pin 153 (B12): device 1e660000.ethernet function RMII1 group RMII1
pin 154 (C12): device 1e660000.ethernet function RMII1 group RMII1
pin 155 (D12): device 1e660000.ethernet function RMII1 group RMII1
pin 156 (E12): device 1e660000.ethernet function RMII1 group RMII1
pin 157 (A13): device 1e660000.ethernet function RMII1 group RMII1
pin 164 (E11): device 1e660000.ethernet function RMII1 group RMII1
pin 165 (D11): device 1e660000.ethernet function RMII1 group RMII1
pin 166 (C11): device 1e660000.ethernet function RMII1 group RMII1
pin 167 (B11): device 1e660000.ethernet function RMII1 group RMII1
pin 168 (A11): device 1e660000.ethernet function RMII1 group RMII1
pin 169 (E10): device 1e660000.ethernet function RMII1 group RMII1
```

amboar, 2017-06-30:
The palmetto problem should be resolved with http://patchwork.ozlabs.org/patch/782661/

amboar, 2017-06-30:
As for barreleye, it cannot be seeing the same problem as there are no FSI nodes in the devicetree. If there are troubles on barreleye please file a different issue.

rfrandse, 2017-07-05:
<https://gerrit.openbmc-project.xyz/5090> kernel: uart, led and gpio fixes
Resolves:  #1836 [Palmetto] fsi-master-gpio: probe of gpio-fsi failed with error -16

closed 2017-07-05 by williamspatrick