openbmc/openbmc #1837: [Palmetto]Insufficient hardware to continue.

swe12345, 2017-06-23:
Insufficient hardware to continue.
Palmetto system reached Quiesce state.
```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```
Driver Details
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-29"
VERSION_ID="v1.99.7-29-ga20607b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-29"
BUILD_ID="v1.99.7"
```
```
root@palmetto:~# cat /etc/timestamp
20170622092455
```

[insufhardware.txt](https://github.com/openbmc/openbmc/files/1096934/insufhardware.txt)

swe12345, 2017-06-23:
[journalctl_23jun.txt](https://github.com/openbmc/openbmc/files/1097362/journalctl_23jun.txt)

williamspatrick, 2017-06-23:
Can you expand on why you think the BMC is to blame for a Hostboot error?

geissonator, 2017-06-26:
Please take this one up with hostboot team, BMC is not involved here.

closed 2017-06-26 by geissonator