openbmc/openbmc #1844: REST API access to BootProgress sensor can take 90+ seconds

stewart-ibm, 2017-06-26:
On a Witherspoon system:

```
root@spoon1-bmc:~# cat /etc/os-release  
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-1"
VERSION_ID="v1.99.6-1-g756dfa6"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-1"
BUILD_ID="v1.99.6"
```

```
stewart@ka3:~/skiboot$ time http_proxy="" https_proxy="" HTTP_PROXY="" HTTPS_PRO
XY="" curl   -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : 
[]}'  -X GET https://spoon1-bmc///org/openbmc/sensors/host/BootProgress

{
  "data": {
    "description": "Login required"
  }, 
  "message": "401 Unauthorized", 
  "status": "error"
}
real    1m32.774s
user    0m0.120s
sys     0m0.032s
```

I'd expect to fail a lot quicker than this. It appears that this is reproducible immediately after having issued power off.

stewart-ibm, 2017-06-26:
```
stewart@ka3:~/skiboot$ time http_proxy="" https_proxy="" HTTP_PROXY="" HTTPS_PROXY="" curl   -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X GET https://spoon1-bmc///org/openbmc/sensors/host/BootProgress
{
  "data": {
    "error": 0, 
    "units": "", 
    "value": "Off"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
real    1m59.745s
user    0m0.132s
sys     0m0.048s
```

stewart-ibm, 2017-06-26:
The heap size for the phosphor-log-manager process is 178671616. This is probably not ideal.

However, if you kill -9 it, the next API call fails:
```
curl   -b cjar   -k  -H  'Content-Type: application/json'   -d '{"data" : []}'  -X GET https://spoon1-bmc///org/openbmc/sensors/host/BootProgress
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2273  100  2260    0    13     52      0  0:00:43  0:00:42  0:00:01   578
{
  "data": {
    "description": "Internal Server Error", 
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)", 
    "traceback": [
      "Traceback (most recent call last):", 
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle", 
      "    route, args = self.router.match(environ)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 935, in custom_router_match", 
      "    route.callback._setup(**args)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 90, in _setup", 
      "    return self.setup(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 395, in setup", 
      "    request.route_data['map'] = self.find(path, callback)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 386, in find", 
      "    path=path)}", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 123, in try_mapper_call", 
      "    return f(**kw)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 60, in get_object", 
      "    retries, interval)", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 42, in retry", 
      "    return func()", 
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 59, in <lambda>", 
      "    path, interfaces, signature='sas'),", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__", 
      "    **keywords)", 
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking", 
      "    message, timeout)", 
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
    ]
  }, 
  "message": "500 Internal Server Error", 
  "status": "error"
}
```

oh, and phosphor-mapper now uses 100% CPU.

bradbishop, 2017-06-26:
The REST server uses systemd socket activation.  So the first connection takes awhile.  If you run it again after the first one does it go any faster?

In any event, the typical start up time for the rest server is closer to 5 seconds so I think there is something else consuming cycles causing the long delay....you mentioned mapper @ 100% - any chance that was already doing that before you killed the error log server?

I'll repro and post back.

stewart-ibm, 2017-06-27:
Brad Bishop <notifications@github.com> writes:
> The REST server uses systemd socket activation.  So the first
> connection takes awhile.  If you run it again after the first one does
> it go any faster?

While a subsequent request is faster, the inital request is the login,
so it's not the first request the server sees (and the BMC itself had
been up for 3+ days). It seems to be related to the power off command
though, it's slow after that, as you can reproduce without restarting
anything on the BMC.

> In any event, the typical start up time for the rest server is closer
> to 5 seconds so I think there is something else consuming cycles
> causing the long delay....you mentioned mapper @ 100% - any chance
> that was already doing that before you killed the error log server?

I don't think so... the REST server would respond to things... just
slowly, then quickly, then slowly.

> I'll repro and post back.

cheers.

A quick glance at the log-mapper code doesn't make me confident that it
doesn't leak memory if you keep posting esels to the BMC.

-- 
Stewart Smith
OPAL Architect, IBM.

bradbishop, 2017-06-27:
I opened https://github.com/openbmc/openbmc/issues/1850 to track the memory leak in phosphor-log-manager.  Lets use this bug to address the mapper busy loop.

@anoo1 Is there a way to fill up the BMC with esels to try and reproduce the high memory usage?

mine260309, 2017-06-27:
@bradbishop If we want to make the BMC with high memory usage, we can create a ramdisk and fill it with dummy data.
E.g.
```
mkdir -p /mnt/tempdir
mount -t tmpfs -o size=512M tmpfs /mnt/tempdir
dd if=/dev/zero of=dummyFile bs=1M count=300
```
Create a ramdisk with size 512M, and we fill it with 300M dummy data, so BMC's free memory is reduced by 300M.
Then we can try to reproduce the issue.

closed 2019-04-27 by stale[bot]