openbmc/openbmc #1845: Kernel corrupts FFS header

cyrilbur-ibm, 2017-06-26:
Under unknown circumstances writes to the host PNOR through the BMC MTD layer have a corrupt first byte.

Three seperate userspace tools produce the same result: `pflash -E -p witherspoon.pnor`, `dd if=witherspoon.pnor of=/dev/mtd7 bs=4096 count=1` and `flashcp witherspoon.pnor /dev/mtd7` all show corruption of the first byte of flash.

The first word of flash after running all three tools is `00000000  d0 41 52 54 | .ART` the expect output of the beginning of an FFS header which would read "PART"

It doesn't seem like there is any other corruption.

Furthermore:
`pflash -e -s 0x1000` to erase the first 4K of the flash and reading it back with `dd if=/dev/mtd7 of=/tmp/header bs=4096 count=1` does show that the entire first 4K of flash is all 0xff.

```
Linux 4.7.10-87950053ee336e99ea0f081695b314c962a0859c #1 Tue Jun 6 14:22:28 CDT 2017 armv6l GNU/Linux

ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.6"
VERSION_ID="ibm-v1.99.6-0-r1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.6"
BUILD_ID="ibm-v1.99.6-0-r1"

dev:    size   erasesize  name
mtd0: 02000000 00010000 "bmc"
mtd1: 00060000 00010000 "u-boot"
mtd2: 00020000 00010000 "u-boot-env"
mtd3: 00440000 00010000 "kernel"
mtd4: 01740000 00010000 "rofs"
mtd5: 00400000 00010000 "rwfs"
mtd6: 02000000 00010000 "alt"
mtd7: 08000000 00001000 "pnor"
```

shenki, 2017-06-26:
Dupe of https://github.com/openbmc/openbmc/issues/1845

williamspatrick, 2017-06-26:
@shenki This is #1845.

geissonator, 2017-06-26:
Dup of #1638 if a power cycle "fixes" the corruption.

shenki, 2017-06-26:
Yeah. #1638 is what I meant :)

closed 2017-06-26 by williamspatrick