openbmc/openbmc #1847: PNOR: Version activation shouldn't be implemented until the next reboot

saqibkh, 2017-06-26:
During activation check for power on. If the Host is up then wait for the next reboot to switch to user requested activation version.

saqibkh, 2017-06-29:
- [ ] If Host is up then create/copy the files to the required dir, but wait to switch to user requested activation version.
- [ ]  If Host is down, then create/copy the files to required dir and switch to user requested activation version.

geissonator, 2017-06-29:
Need to handle reset/reload - go back to previous image

rfrandse, 2017-07-18:
<https://gerrit.openbmc-project.xyz/5336> PNOR: Version activation shouldn't be implemented until the Host is powered down.
Resolves  #1847 PNOR: Version activation shouldn't be implemented until the next reboot

closed 2017-08-18 by williamspatrick